shanteacontrols/OpenDeck #5: led limit

PierreKircher, 2017-03-14:
the led limit would be gone if you use ws2812b they have basicaly 1 dataline as communication bus they work fine with the 32u4

paradajz, 2017-03-14:
Hardware limit - yes. Software one - not quite. I've used nearly all RAM on 32u4 for current firmware. Also I don't have a single pin available anymore, so this won't be possible on current hardware revision. In the future I'm going to switch to either ARM (STM32) or bigger AVR (ARM is more likely).

PierreKircher, 2017-03-14:
arm cortex m4  might be a good option .. plenty mem / and flash .. and 180 mhz !

paradajz, 2017-03-14:
Indeed, I've played around with STM32F4 series and they're quite capable. Only problem is that for OpenDeck I'm using LUFA as USB framework since I don't feel like messing with USB stuff yet, and there isn't similar alternative for ARM, so I'd have to write that myself. That's pretty much the only reason I don't use ARM at the moment.

PierreKircher, 2017-03-14:
hmm vusb is a other option ( yaya i hear ya) .. but you might be able to talk to haata .. (input club) ergodox infinity .. they use a cortex platform for the keyboard .. and afaik lufa as well

PierreKircher, 2017-03-15:
thats a usb hid (keyboard) / midi implementation for the teensy 3.6 (cortex m4)

https://github.com/PaulStoffregen/USBHost_t36 

it might gives some clues / helps at the ARM switch

paradajz, 2017-03-24:
I'll look into it. Thanks!

closed 2017-03-24 by paradajz