shanteacontrols/OpenDeck #6: Feature wishes

Momecmusic, 2018-01-05:
Hi Igor,

thank you for this awesome project, I love my opendeck board and the configurator is super solid.

I have a few wishes that I hope you will consider implementing:

1. The possibility of having transport controls. Start/Stop in particular would be a very nice option to have. 

2. This is a very specific request, one that relates directly to controlling an Elektron Analog Rytm drummachine.
So, my request goes to the possibility of having upper and lower CC limits on the LED's. This is because Analog Rytm has a scene mode function with 12 scenes, activated by each of the 12 pads on the Rytm. Oddly enough, they chose to have all 12 scenes controlled on the samme CC value, namely 92, and then being able to trigger each scene with the velocity of CC92. Therefore, if I want to make a controller with buttons for the Scene modes, I have to use the analog inputs on the OpenDeck, which is fine since we have the CC limit controls, but if I want to activate a separate LED for each button this is not yet possible.

I hope this makes sense and that you will consider my suggestions.

And thanks again for this project.

Best,

Morten

paradajz, 2018-01-05:
Hi Morten,

glad you like it.

1. Do you want the transport controls to be mappable to buttons or something else?
2. It's clear to me what Elektron does, but I'm not sure what do you want to do with OpenDeck. I assume you are controlling those scenes using single potentiometer, am I right? And then you want to control LEDs on OpenDeck locally using the CC controller value?

Momecmusic, 2018-01-05:
Hi Igor, thank you for the very fast response.

1. Yes exactly, mappable to buttons so that I can trigger start/stop from the midi controller. 

2. I can see that I did not explain myself well.

I want to trigger the scenes with backlit pushbuttons on the midi controller. So, button1 will send out CC92 at velocity 1, button2 will send CC92 at velocity 2 and so on.

And yes, I then want to control the LEDs locally, but be able to limit each LED for the pushbuttons to the corresponding velocity if this is at all possible. As it is now, all LEDs will fire when pushing any button.

Hope this makes sense, otherwise let me know and I will try to explain further.

Best,

Morten

paradajz, 2018-01-05:
1. I'll try to implement it by the end of the week - it shouldn't take much work. I'll add support for Play, Stop, Pause and Record with selectable MIDI ID/Channel number (see: https://en.wikipedia.org/wiki/MIDI_Machine_Control#MMC_Message_Format)

2. I need to take a closer look into this since it will require some changes and I want to preserve compatibility with older versions.

Momecmusic, 2018-01-05:
Very nice, no rush with this but sooner is always better of course.

I completely understand regarding my second point. This issue is also a bit weird and only relates to the Rytm. I have not yet seen a Midi controller that let's you control the LEDs like this, so I know it is an unusual situation.

But thank you so much for looking in to these two issues.

Best,

Morten

paradajz, 2018-01-05:
No problem.

By the way - are you connecting buttons to analog inputs configured as potentiometers to send those CCs?

Momecmusic, 2018-01-05:
Yes, exactly.

Would also be nice to be able to have CC limits when connected as buttons, if possible, but it works great when connected to the analog ins as I do now. Of course we don't have the Latching/Momentary option with analog ins, but for me personally that is a not a problem.

paradajz, 2018-01-05:
Yeah that looks like ugly hack, though. Will look into that as well.

Momecmusic, 2018-01-05:
Ha ha, I am a coding and Midi noob, but I could imagine that being a nightmare :)

paradajz, 2018-01-05:
First issue is now implemented https://github.com/paradajz/OpenDeck/commit/47cdfb912a4caf1fd2e3ba9b2a48a37bc68c218b
Can't test it at the moment but it should work fine. Web interface has also been updated.

Momecmusic, 2018-01-05:
Wow, damn you are fast :)

Thank you!

paradajz, 2018-01-07:
I did some more changes:
Now, velocity/control value is configurable for every button. So, if you configure button to send CC messages, control value (third byte) is selectable, so you don't have to use that hacky way of connecting button to analog input configured as potentiometer. I've also added two different CC behaviours for buttons - first is same as before: CC with selected velocity/control value is sent on press, and that same CC value is sent with control value/velocity 0 on release. Second one is the same, but nothing happens once button is released. This option depends on button type (momentary or latching) so you can play around with these settings until you get wanted behaviour. As for the LEDs - how exactly have you wanted to turn them off?

paradajz, 2018-01-07:
I think I've now actually solved the issue. Basically, I simply introduced another configurable parameter for LEDs: activation velocity, so now you can select both activation note and velocity when controlling LEDs. Any velocity which isn't the one specified will turn the LED off. This velocity is used only for single color LEDs, for RGB LEDs velocity table is used as before. If I configure setup like this:

Button 0: CC 92, value 127, no reset to 0
Button 1: CC 92, value 126, no reset to 0
LED 0: Activation ID: 92, activation velocity 127, local LED control
LED 0: Activation ID: 92, activation velocity 126, local LED control

Pressing button 0 will turn LED 0 on.
Pressing button 1 will turn LED 0 off and turn LED 1 off.
Pressing button 0 again will turn LED 1 off and turn LED 0 back on.

You get the idea...

I need to test it some more and update the documentation but I will probably release it tomorrow.

Momecmusic, 2018-01-08:
Wow, I am really impressed.

These solutions you came up with sounds exactly like what I need, and you even came up with some other cool options for other scenarios.

And, being able to use buttons instead of the analog ins is really nice. Great work.

The solution for the LEDs again sounds exactly right.

Can't wait to test these new features out.

Oh, and thank you again for implementing transport functions, I can now build a controller exactly like I hoped to be able to, since I could not find anything already existing that would work. 

I believed in this project from the beginning, but now I am just all in and will recommend it to everyone.

paradajz, 2018-01-08:
Thanks, means a lot! Until I update documentation, you can download current firmware here:
http://shanteacontrols.com/dev/FLASH.bin
Let me know if you hit any bugs.

Momecmusic, 2018-01-08:
Cool, thanks. Unfortunately I have a lot of work the next couple of days, but will test a.s.a.p.

Momecmusic, 2018-01-08:
Hi again,

I brought the board with me to work to test the new firmware. Updating the board worked fine, it boots up as it should. However I do not see any of the new features in the Web configurator yet.

Maybe I made a mistake during the firmware update. Is there any way to check that? Pretty simple process though, so I am quite confident it went ok.

paradajz, 2018-01-08:
Try clearing Chrome cache - as I said sometimes it loads old version even after the update.

Momecmusic, 2018-01-08:
Doh! Should have thought of that :)

Thanks, works now.

paradajz, 2018-01-08:
Great. I'll close this for now, if you find any bugs or if you think of any other feature, please open new ticket.

closed 2018-01-08 by paradajz