shanteacontrols/OpenDeck #7: MMC Messages

Momecmusic, 2018-01-24:
Hi again Igor,

so I have now testet all of the new functions and everything is great and works really well, except the MMC messages.

I can not get the Analog Rytm to accept any Start/stop action.

I have used the Arturia Beatstep pro for a long time to sync and control the Analog Rytm, and this is the message that it sends out:
[image]

The OpenDeck sends out this:
[image]

I know that the MMC send by OpenDeck is right and should work, but I don't know why the Beatstep Pro works and what type of message that is.
The Analog Rytm manual does not mention anything about the MMC implementation in the MIDI appendix, but I could try to contact Elektron support in case you don't know what's going on either.

Do you have any insight in to this matter?

Best,

Morten

paradajz, 2018-01-24:
In MIDI, there are single-byte real-time commands such as Play (0xFA), Stop (0xFC), Clock (0xF8) etc.. See [here](https://github.com/paradajz/midi-lib/blob/master/src/DataTypes.h#L42). What I don't understand why are there single-byte Start/Stop commands and SysEx-like Play/Stop MMC commands. At the moment, only MMC messages are supported in OpenDeck, but whatever, I'll add support to map buttons to those single-byte commands as well later today.

Momecmusic, 2018-01-24:
Cool, thank you.

Well, I don't really understand it either and the documentation on this from Elektron's side is non-existent.

You are the man!

paradajz, 2018-01-24:
OK so here's the updated firmware (beta - I can't test it right now):
http://shanteacontrols.com/dev/FLASH.bin

Also make sure to clean your Chrome cache since I've updated WebUI as well.

Momecmusic, 2018-01-24:
Yes! It works. Have not testet every function of the new firmware, but the real time functions are there and working and I expect everything else to be fine as well.

Once again I can not express my gratitude, I am a very happy customer.

I am expecting the last parts to arrive tomorrow and will shoot some photos for you of my setup this week or early next week.

You can close this ticket now, I have no further requests :)

paradajz, 2018-01-24:
Great. Also, between this version and last one, I've added support for 7-bit and 14-bit NRPN, so you can test that as well if you need it.

paradajz, 2018-01-25:
Additional fix: Added real-time messages weren't sent correctly using USB interface. Referenced commit:
https://github.com/paradajz/midi-lib/commit/d533c0d3b6ea26c062dfb38d94ea772ce28560fa

Also, make sure to download FLASH.bin again - I've sent you firmware compiled for older revision of OpenDeck board which I'm using for development. That board has different analog component mapping, so make sure you update the firmware (in web interface, board version should be 1.2.0).

Momecmusic, 2018-01-26:
Ok, thanks Igor.

Don't think I will need NRPN personally, but it is great to have that option. As far as I understand it, NRPN is more data heavy? So it will be slower? Or are there any benefits I am missing?

Momecmusic, 2018-01-26:
Hi again,

web interface says that board version is 1.4 now.

paradajz, 2018-01-26:
Can you screenshot this please?

Momecmusic, 2018-01-26:
Of course:

[image]

paradajz, 2018-01-26:
Weird. I'll take a look at it.

paradajz, 2018-01-26:
Regarding NRPN, the point is so have more than 127 CC messages, and also to allow 14-bit values for those same CC messages. Standard CC message is one byte long, 7-bit NRPN is 3 bytes and 14-bit NRPN is 4 bytes, so yes, it involves more traffic. You can take a look at it here:

http://www.2writers.com/eddie/TutNrpn.htm

Momecmusic, 2018-01-26:
Arh, of course, thanks. Hmmm, might play around with it a bit to see what works best.

paradajz, 2018-01-29:
Web UI should now report hw/sw info correctly (if firmware is updated to v1.5.0).

Momecmusic, 2018-01-29:
Thanks,

not sure I have v1.5.0 though. Still says 1.4 in the Web UI, and I can only find v1.2 available (By clicking "Check for updates) link in Web UI

paradajz, 2018-01-29:
Have you cleared your cache? This is how it looks here:

[image]

Checking for release should work fine as well, v1.5.0 is latest:
https://github.com/paradajz/OpenDeck/releases

paradajz, 2018-01-29:
BTW, v1.2.0 inside release archive is not firmware version, but hardware version. Since you have board v1.2.0, you need file in bin/compiled/board/opendeck/v1.2.0/FLASH.bin. That's firmware v1.5.0

closed 2018-01-29 by paradajz

Momecmusic, 2018-01-29:
Yes, 

I cleared my cache. 

Just now when I went to the Web UI I am experiencing some issues. 

Midi page is blank, and it reports "ERROR INDEX" in the activity section.

The other pages work, and I can program the board, but when I click in Info, it just stalls at "Searching for OpenDeck board".

paradajz, 2018-01-29:
I have sent you an email - let's continue this discussion there as it isn't related to this ticket.