shanteacontrols/OpenDeck #9: Can't build atmega16u2 bootloader for Arduino Mega 2560

IrregularShed, 2018-07-18:
Hi!

I've been trying to build and flash the bootloader for an Arduino Mega. I'm using the Windows Subsystem for Linux and have built the firmware for a Pro Micro already, which works perfectly. I've also built and flashed the `fw_mega` firmware, manually from a Windows PowerShell prompt and with the `arduino_flash.sh` script, both with my Arduino Uno as ISP and with my Bus Pirate. All that works as I would expect.

I've had problems with the bootloader for the 16u2 though, and the Arduino is completely unreachable from any computer when I plug it in now. (My Linux machine doesn't even show a system event when I plug it in!) I tried updating the code on my machine to build from master but it's still not working. I get different errors depending on what I try:
```
steve@LAPTOP-35A1V7S5:~/OpenDeck/src$ make boot_16u2
Cleaning up.
Building target boot_16u2
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Template/Template_Endpoint_Control_R.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Template/Template_Endpoint_Control_W.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Template/Template_Endpoint_RW.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Template/Template_Pipe_RW.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/Events.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/USBTask.c
In file included from bootloader/hid/BootloaderHID.c:36:0:
bootloader/hid/BootloaderHID.h:63:45: fatal error: board/avr/variants/xu2/Hardware.h: No such file or directory
compilation terminated.
make: *** [build/bootloader/hid/BootloaderHID.o] Error 1
```
... so something is missing when building with `make`.

```
steve@LAPTOP-35A1V7S5:~/OpenDeck/scripts$ sudo ./arduino_flash.sh
Please type in correct tty port on which ArduinoISP is connected (ie. ttyUSB0):
ttyS8
Please select MCU you want to flash and then press enter:
1 - ATmega8u2 on Uno or Mega
2 - ATmega16u2 on Uno or Mega
3 - ATmega328P on Uno
4 - ATmega2560 on Mega
5 - ATmega32u4 on Leonardo
6 - ATmega32u4 on Pro Micro
2
Connect programmer to ATmega16u2 programming header on the board and then press enter.

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9489
avrdude: erasing chip
avrdude: reading input file "0xff"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.12s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xff:
avrdude: load data lock data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x3f != 0xff
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (H:F8, E:D3, L:FF)

avrdude done.  Thank you.

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9489
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "../src/build/fw_16u2.bin"
avrdude: error opening ../src/build/fw_16u2.bin: No such file or directory
avrdude: input file ../src/build/fw_16u2.bin auto detected as invalid format
avrdude: can't open input file ../src/build/fw_16u2.bin: No such file or directory
avrdude: read from file '../src/build/fw_16u2.bin' failed

avrdude: safemode: Fuses OK (H:F8, E:D3, L:FF)

avrdude done.  Thank you.
```
... so verification is failing there (ignoring the issue with the firmware file missing).

Any help will be appreciated!

IrregularShed, 2018-07-19:
Just tried building fw_16u2 - everything worked (although the configurator is a bit hit-and-miss connecting to it). I'll close this now, and have a play!

(btw I've got avrdude 6.3 in a Windows PowerShell, 6.0.1 in WSL.)

closed 2018-07-19 by IrregularShed

paradajz, 2018-07-19:
OK great - yeah configurator is sometimes weird when opening a connection.

Regarding the avrdude - I'm using v6.3 in WSL (Ubuntu 18.04). Works without throwing that verification error.

You should also reflash the firmware using latest commit since I've fixed some stuff this morning (incorrect button order, indicator LEDs etc.)

IrregularShed, 2018-07-19:
:+1: will do!

Quick question, is it possible to add midi ports to Mega? I've got a USB-to-midi interface that doesn't run with any modern version of Windows and I was thinking about pulling it all apart to hack. (I know I should put this in a new issue...)

Jozio611, 2024-02-02:
Hello. I have a problem with 16u2, the computer detects it as OPENDECK 16u2 Dfu. my system is windows 10. please help. Thank you