shanteacontrols/OpenDeck #10: Arduino ProMicro version reports 10 buttons

mungewell, 2018-07-19:
Documentation says it supports 6 buttons, but when queried it reports 10
https://github.com/paradajz/OpenDeck/wiki/Connections#arduino-pro-micro
```
$ amidi -p hw:1,0,0 -S 'F0 00 53 43 00 00 42 F7' -r resp.bin ; hexdump -Cv resp.bin 
^C
11 bytes read
00000000  f0 00 53 43 01 00 03 01  00 00 f7                 |..SC.......|
0000000b
$ amidi -p hw:1,0,0 -S 'F0 00 53 43 00 00 4D F7' -r resp.bin ; hexdump -Cv resp.bin 
^C
11 bytes read
00000000  f0 00 53 43 01 00 0a 03  04 06 f7                 |..SC.......|
0000000b
```

Physically the first button (pin 2) does not function, however pin3 reports as 'button5' and when configuration is changed (via WebUI) the changes to button5 are reflected in the Midi stream.

paradajz, 2018-07-19:
Please report two different issues next time instead of combining into one.

Reported number is fine, since on OpenDeck supports conversion of analog inputs to digital ones, therefore, maximum number of buttons is reported as maximum number of digital inputs combined with maximum number of analog inputs. In Pro Micro case, maximum amount of digital inputs is 6, and maximum amount of analog inputs is 4 - hence the reported total number 10.

It is vaguely mentioned here:
https://github.com/paradajz/OpenDeck/wiki/Configurable-features#type-section-1

And in SysEx page I've fixed the info:
https://github.com/paradajz/OpenDeck/wiki/SysEx-Configuration#number-of-supported-components

As for the pin 2 not functioning, I've just fixed it (https://github.com/paradajz/OpenDeck/commit/548d69db52bfb189a86e85eda4fd3638c99fde14).

Also, LED indicators (for MIDI traffic) now work properly on Pro Micro (https://github.com/paradajz/OpenDeck/commit/3353ee49124f99fca5cccd0b9de02188deea545e).

Bugs to be expected on master :)

closed 2018-07-19 by paradajz

mungewell, 2018-07-20:
Thanks for being so responsive/fast on the fixes :-) Seems to be working correctly now.