shanteacontrols/OpenDeck #11: Arduino Pro Micro - WebUI Midi Tab report Index Error

mungewell, 2018-07-20:
When switching to the Midi tab, the monitor show 'Index Error' multiple times and there are no controls shown/given. Other tabs appear to functioncorrectly...

Screenshot attached.
[image]

I've switched on MIDI DIN (via command line) and these are the messages output when switching to the MIDI tab - maybe they are a hint?
```
$ hexdump -Cv minicom.cap
00000000  28 32 30 08 20 08 29 0a  52 61 77 20 55 41 52 54  |(20. .).Raw UART|
00000010  20 69 6e 70 75 74 0a 41  6e 79 20 6b 65 79 20 74  | input.Any key t|
00000020  6f 20 65 78 69 74 0a f0  53 43 01 0a 03 04 06 f7  |o exit..SC......|
00000030  f0 53 43 01 f7 f0 53 43  01 f7 f0 53 43 01 f7 f0  |.SC...SC...SC...|
00000040  53 43 01 01 f7 f0 53 43  09 f7                    |SC....SC..|
0000004a
```

paradajz, 2018-07-20:
You are using older version of the interface - latest version is located here (hosted on GitHub):
https://paradajz.github.io/OpenDeck/#/

It's linked via shanteacontrols.com and mentioned in README.md

closed 2018-07-20 by paradajz

mungewell, 2018-07-20:
Thanks for the new link, however the I had followed a/the link on the main page (Readme.md??)
[image]

paradajz, 2018-07-20:
You're right - it's linked twice in README and the first link was incorrect - fixed!