shanteacontrols/OpenDeck #12: configurator- analog

quirosaur, 2018-08-08:
only first analog port works. in configurator, if i enable say, 1 and 4 then when moving potentiometer in A0, configurator flashes both 1 and 4(same goes for signal screen). but when i try to use any other potentiometer in those other ports(A1-A5), nothing happens.

ps. i have multiplexers, can i use them to increase amount of things i can attach(i.e. are they supported?)

paradajz, 2018-08-08:
Hi,

1) Which board are you using?
2) Which firmware are you running?

quirosaur, 2018-08-08:
Also, so that i won't forget: there was problem when compiling the firmware file:these { } brackets were missing near one if clause, so i added them, and it compiled succesfully. 

https://github.com/paradajz/OpenDeck/blob/master/src/application/board/avr/variants/ISR.cpp

row 119

(compiler threw error on me)

paradajz, 2018-08-08:
On the quick glance on the relevant code I'm not seeing anything suspicious. Note that you must enable only the analog inputs on which you have analog inputs connected. Otherwise, noise/crosstalk (effect that moving potentiometer registers as moving some other pot) will occur.

Regarding the code compilation - I'm not getting any errors (and neither does the build system on Travis) - which distribution/compiler are you using?

paradajz, 2018-08-08:
Actually you are right - ugh. What a horrible bug. Should be fixed now:

https://github.com/paradajz/OpenDeck/commit/92b8ee7404c5ec0da7bec8af72cec351839b6d7e

quirosaur, 2018-08-08:
Hi, i followed the tutorial on osx. so homebrew, uso as avr programmer and leonardo as target board. i tested analog using rotary potentiometers, but messages flooded and crashed the configurator. sliding potentiometer works well, only problem being that i cannot use more than one.

Just tested with mixx sotware, the one slider and buttons work well. i also plan to follow your tutorial on hdd motors as rotary encoders, i'm only missing the zener diodes atm, but will order those soon as well

quirosaur, 2018-08-08:
glad to be of help :) and also, the end bracket was missing, was it on the line 139?

paradajz, 2018-08-08:
Take a look at the linked commit. The point was that because of the missing bracket analog input was never switched, resulting in constant readout of first input only. Let me know if that fixes the issue.

quirosaur, 2018-08-08:
yeah, i'll check that. Thanks man!

quirosaur, 2018-08-08:
So, i tested flashing after recloning the repository. it had no effect, and same problem persists.
i attached console prints here, if they're any help

```
Cleaning up.
Building target fw_leonardo
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Template/Template_Endpoint_Control_R.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Template/Template_Endpoint_Control_W.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Template/Template_Endpoint_RW.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/Template/Template_Pipe_RW.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/Events.c
Finished building: modules/lufa/LUFA/Drivers/USB/Core/USBTask.c
Finished building: application/board/avr/usb/Descriptors.c
Finished building: modules/lufa/LUFA/Drivers/USB/Class/Device/AudioClassDevice.c
Finished building: modules/lufa/LUFA/Drivers/USB/Class/Device/MIDIClassDevice.c
Finished building: application/OpenDeck.cpp
Finished building: modules/core/src/HAL/avr/reset/Reset.cpp
Finished building: application/database/Database.cpp
Finished building: modules/dbms/src/DBMS.cpp
Finished building: application/interface/analog/Analog.cpp
Finished building: application/interface/analog/FSR.cpp
Finished building: application/interface/analog/Potentiometer.cpp
Finished building: application/interface/cinfo/CInfo.cpp
Finished building: application/interface/digital/input/DigitalInput.cpp
Finished building: application/interface/digital/input/buttons/Buttons.cpp
Finished building: application/interface/digital/input/encoders/Encoders.cpp
Finished building: application/interface/digital/output/leds/LEDs.cpp
Finished building: application/sysExConf/Handling.cpp
Finished building: modules/midi/src/MIDI.cpp
Finished building: modules/sysex/src/SysEx.cpp
Finished building: application/board/avr/variants/Common.cpp
Finished building: application/board/avr/variants/ISR.cpp
Finished building: application/board/common/analog/input/Common.cpp
Finished building: application/board/common/digital/input/encoders/Common.cpp
Finished building: application/board/common/digital/input/DigitalIn.cpp
Finished building: application/board/common/digital/output/DigitalOut.cpp
Finished building: application/board/avr/variants/leonardo/HardwareControl.cpp
Finished building: application/board/avr/variants/leonardo/Init.cpp
Finished building: application/board/avr/variants/leonardo/pins/Map.cpp
Finished building: application/board/common/digital/input/buttons/DirectIn.cpp
Finished building: application/board/common/digital/input/encoders/DirectIn.cpp
Finished building: application/board/common/digital/output/leds/DirectOut.cpp
Finished building: application/board/avr/usb/USB_MIDI.cpp
Finished building: application/board/avr/uart/UART.cpp
Finished building: application/interface/midi/OpenDeckFormat.cpp
Finished building: application/interface/midi/STDFormat.cpp
Finished building target: build/fw_leonardo.elf
srec_cat: warning: option "-Little_Endian_MAximum" is deprecated, please use
    "-MAximum_Little_Endian" instead
srec_cat: warning: option "-Little_Endian_Cyclic_Redundancy_Check_16" is
    deprecated, please use "-Cyclic_Redundancy_Check_16_Little_Endian" instead
AVR Memory Usage
----------------
Device: atmega32u4

Program:   22376 bytes (68.3% Full)
(.text + .data + .bootloader)

Data:       1062 bytes (41.5% Full)
(.data + .bss + .noinit)
```
[... 70 lines cut ...]
avrdude: reading input file "0xef"
avrdude: writing lock (1 bytes):

Writing |                                                    | 0% 0.00s ***failed;  
Writing | ################################################## | 100% 0.11s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xef:
avrdude: load data lock data from input file 0xef:
avrdude: input file 0xef contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x2f != 0xef
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FB, H:D8, L:FF)

avrdude done.  Thank you.
```

and flashing gives these mismatch errors, but i'm not sure are those essential.

paradajz, 2018-08-08:
Found another bug with pin switching - it's fixed now (really!).

https://github.com/paradajz/OpenDeck/commit/e1693fa69da3728723e06cc20a692f9a889d9f9b

closed 2018-08-08 by paradajz