shanteacontrols/OpenDeck #14: ceylon/HDD motor encoder

quirosaur, 2018-09-16:
Hi again! 

i'm trying to follow your tutorial on using hdd motor as an encoder. however, i ran in to a couple of problems: according to schematic, it says to use pins 7, 6 and 1 on both inputs (comparing hdd pins 1 and 3/ 2 and 3) is it a mistake, because in text you mention that you needed 2 comparators and lm339 has 4.

Then the second question: in tutorial, you connected comparator output to pins 1 and 2(interrupt pins), but in wiki of opendeck, encoders are connected to 2/3, 4/5 and 6/7.

adding to that, you showed where you made changes to code. is it needed to change and upload code to arduino, when doing this?

thanks, Samu

paradajz, 2018-09-16:
I'm not sure what do you mean with 'mistake'. Could you elaborate? Three outputs from HHD are needed, however, one of those outputs is reference which is being used for both inputs.

That post is really old anyways and currently, OpenDeck doesn't support such fast encoders. I could implement this again on Arduino boards however sometime soon.

quirosaur, 2018-09-17:
Okay, by mistake i meant this: 

[image]

which says that 2 comparators are used out of 4. but:

[image]

shows that only pins 6,7 and 1 are used i.e. only one comparator out of 4.

then, about the pins, i misread. sorry. 

But i understand that these are not supported at the moment, thanks for replying! 

ps. if it isn't big trouble, hdd motors as encoders would be awesome

closed 2018-09-17 by quirosaur

paradajz, 2018-09-17:
_shows that only pins 6,7 and 1 are used i.e. only one comparator out of 4._

Ah, I see now - yeah it's an mistake then. I've actually used pins 4, 5, 6 and 7 on LM339 as inputs. As I've said - that's really outdated. I'll see what I can do about implementing that kind of setup again.