shanteacontrols/OpenDeck #15: Using OpenDeck to forward MIDI data to hardware Synth

mungewell, 2018-12-05:
Hi, I am trying to understand how/whether OpenDeck can be used to forward USB MIDI to it's DIN output, and onto a hardware synth. I want to go:
PC--USB-->OpenDeck--DIN-->Synth

I think that this is related to the MIDI merge interface control:
https://github.com/paradajz/OpenDeck/wiki/SysEx-Configuration#midi-merge

On my Ardunio Pro-Micro hardware I have tried setting this to 'DIN out' (which didn't seem to output) and to 'OpenDeck Master' (which gave error code '0x0A' - guess that's not in the f/w I have on device).

Am I heading in the right direction?
Thanks,
Simon

[opendeck-check.sh.txt](https://github.com/paradajz/OpenDeck/files/2647086/opendeck-check.sh.txt)
[opendeck-din-out.sh.txt](https://github.com/paradajz/OpenDeck/files/2647087/opendeck-din-out.sh.txt)
[opendeck-master.sh.txt](https://github.com/paradajz/OpenDeck/files/2647088/opendeck-master.sh.txt)
[opendeck-off.sh.txt](https://github.com/paradajz/OpenDeck/files/2647089/opendeck-off.sh.txt)

```
$ ./opendeck-check.sh 
^C
7 bytes read
00000000  f0 00 53 43 01 00 f7                              |..SC...|
00000007
^C
14 bytes read
00000000  f0 00 53 43 01 00 02 00  03 03 01 00 00 f7        |..SC..........|
0000000e
Read MIDI Merge State
^C
8 bytes read
00000000  f0 00 53 43 01 00 01 f7                           |..SC....|
00000008
Read DIN MIDI State
^C
8 bytes read
00000000  f0 00 53 43 01 00 01 f7                           |..SC....|
00000008
Read MIDI Merge Interface
^C
8 bytes read
00000000  f0 00 53 43 01 00 01 f7                           |..SC....|
00000008
^C
7 bytes read
00000000  f0 00 53 43 01 00 f7                              |..SC...|
00000007
```

paradajz, 2018-12-05:
Hi, no, in Web configurator you only need to enable DIN MIDI and leave merging disabled. Merging is used for when you want to forward incoming DIN MIDI data to DIN MIDI out, USB, or both.

mungewell, 2018-12-05:
Hmmm, I think that was what I had previously seen (with #11) but doesn't seem to work now.

Either I am being an idiot, or perhaps my hardware is faulty. The current hardware was not working with the 'general' tab in the configurator and I was having difficulty reflashing it. The current f/w revision was reported as
```
00000000  f0 00 53 43 01 00 02 00  03 03 01 00 00 f7        |..SC..........|
0000000e
```

I'll try to flash another Ardunio and see if that works for me.
Thanks.

paradajz, 2018-12-05:
That tab doesn't work anymore with any firmware other that >3.1.0 as explained in the blog post:
https://shanteacontrols.com/2018/11/19/opendeck-sw-v3-1-0-presets/

paradajz, 2018-12-05:
Also note that Arduino boards require certain resistors for DIN MIDI in/out as mentioned here:
https://github.com/paradajz/OpenDeck/wiki/Hardware#arduino-pro-micro-3

closed 2018-12-05 by paradajz

Potreba, 2023-03-05:
> Also note that Arduino boards require certain resistors for DIN MIDI in/out as mentioned here: https://github.com/paradajz/OpenDeck/wiki/Hardware#arduino-pro-micro-3

Hi. Sorry for reviving old thread, but is there a way to open older wiki and learn how to add DIN midi to these older boards? Unfortunately this links redirects to Wiki home page.  I found no schematic with google search as to where and how should I connect DIN out port.