shanteacontrols/OpenDeck #21: Control LED brightness

tschiggi451, 2020-01-29:
I'd like to MIDI control the brightness of single color & RGB LED (e.g. 25%, 50%, 75% and 100%). This would be super useful to program button states, e.g. off / disabled / enabled / active.

BTW. Thanks a lot for starting this great project!

paradajz, 2020-01-29:
I am thinking about the best way to implement this while staying compatible with the current system:

https://github.com/paradajz/OpenDeck/wiki/LED-control

What I had in mind was:

1) For separate state/blink messages the current values are:

| Velocity/control value/program range | Color   |
|--------------------------------------|---------|
| 0-15                                 | Off     |
| 16-31                                | Red     |
| 32-47                                | Green   |
| 48-63                                | Yellow  |
| 64-79                                | Blue    |
| 80-95                                | Magenta |
| 96-111                               | Cyan    |
| 112-127                              | White   |

In order to implement brightness control, I could simply scale the defined range for specific color to brightness, ie. for red, the values would be something like this:

| Value | Brightness   |
|--------------------------------------|---------|
| 16-20                                | 50%     |
| 21-25                                | 75%     |
| 26-31                                | 100%     |

Fits into existing model quite well.

Where this wouldn't work is when system for combined state and blinking is used, ie:
| Velocity/control value | Color    | Blink speed |
|------------------------|----------|-------------|
| 0-15                   | Off      | No blinking |
| 16                     | Red      | 100ms       |
| 17                     | Red      | 200ms       |
| 18                     | Red      | 300ms       |
| 19                     | Red      | 400ms       |
| 20                     | Red      | 500ms       |
| 21                     | Red      | 600ms       |
| 22                     | Red      | 700ms       |
| 23                     | Red      | 800ms       |
| 24                     | Red      | 900ms       |
| 25                     | Red      | 1000ms      |
| 26-31                  | Red      | No blinking |
| 32                     | Green    | 100ms       |
| 33                     | Green    | 200ms       |
| 34                     | Green    | 300ms       |
| 35                     | Green    | 400ms       |
| 36                     | Green    | 500ms       |
| 37                     | Green    | 600ms       |
| 38                     | Green    | 700ms       |
| 39                     | Green    | 800ms       |
| 40                     | Green    | 900ms       |
| 41                     | Green    | 1000ms      |
| 42-47                  | Green    | No blinking |
| 48                     | Yellow   | 100ms       |
| 49                     | Yellow   | 200ms       |
| 50                     | Yellow   | 300ms       |
| 51                     | Yellow   | 400ms       |
| 52                     | Yellow   | 500ms       |
| 53                     | Yellow   | 600ms       |
| 54                     | Yellow   | 700ms       |
| 55                     | Yellow   | 800ms       |
| 56                     | Yellow   | 900ms       |
| 57                     | Yellow   | 1000ms      |
| 58-63                  | Yellow   | No blinking |
| 64                     | Blue     | 100ms       |
| 65                     | Blue     | 200ms       |
| 66                     | Blue     | 300ms       |
| 67                     | Blue     | 400ms       |
| 68                     | Blue     | 500ms       |
| 69                     | Blue     | 600ms       |
| 70                     | Blue     | 700ms       |
| 71                     | Blue     | 800ms       |
| 72                     | Blue     | 900ms       |
| 73                     | Blue     | 1000ms      |
| 74-79                  | Blue     | No blinking |
| 80                     | Magenta  | 100ms       |
| 81                     | Magenta  | 200ms       |
| 82                     | Magenta  | 300ms       |
| 83                     | Magenta  | 400ms       |
| 84                     | Magenta  | 500ms       |
| 85                     | Magenta  | 600ms       |
| 86                     | Magenta  | 700ms       |
| 87                     | Magenta  | 800ms       |
| 88                     | Magenta  | 900ms       |
| 89                     | Magenta  | 1000ms      |
| 90-95                  | Magenta  | No blinking |
| 96                     | Cyan     | 100ms       |
| 97                     | Cyan     | 200ms       |
| 98                     | Cyan     | 300ms       |
| 99                     | Cyan     | 400ms       |
| 100                    | Cyan     | 500ms       |
| 101                    | Cyan     | 600ms       |
| 102                    | Cyan     | 700ms       |
| 103                    | Cyan     | 800ms       |
| 104                    | Cyan     | 900ms       |
| 105                    | Cyan     | 1000ms      |
| 106-111                | Cyan     | No blinking |
| 112                    | White    | 100ms       |
| 113                    | White    | 200ms       |
| 114                    | White    | 300ms       |
| 115                    | White    | 400ms       |
| 116                    | White    | 500ms       |
| 117                    | White    | 600ms       |
| 118                    | White    | 700ms       |
| 119                    | White    | 800ms       |
| 120                    | White    | 900ms       |
| 121                    | White    | 1000ms      |
| 122-127                | White    | No blinking |

In this case I would have to use another message for brightness, which gets kinda confusing, so you either have:

1) State/Color with brightness message + blink message
or
2) State/Color/Blink message + brightness message

Not sure how to deal with this.

tschiggi451, 2020-01-31:
Hi @paradajz ,

Thanks for your reply!

Your proposed solution would totally work for me (if 25% would be an option as well ;) ). Not offering brightness support when blinking seems a valid option. Another solution I could imagine would be offering brightness information as optional via MIDI channel 2. If no information is send, the brightness would stay at 100%, just as it is today.

Thanks and all the best,

Jonas

paradajz, 2020-01-31:
The thing is, you can configure MIDI channel for each LED individually, so hardcoding brightness against MIDI channel 2 wouldn't really work. I'll think some more about this.

paradajz, 2020-06-03:
Will start implementing this soon, however it will not be available on the current OpenDeck board - only on the next one (v2). I've also decided not to implement brightness control while the LED is blinking.

tschiggi451, 2020-06-04:
Cool, thanks for letting me know. Looking forward!

Best,
Jonas

paradajz, 2020-12-01:
With the [v5.1.0](https://github.com/paradajz/OpenDeck/releases/tag/v5.1.0) release, I've added initial infrastructure for handling LED brightness, although the actual brightness control still isn't implemented. Hoping to get it done in next release. I've also completely reworked control values, so you can check the new control table here (spoiler alert: brightness control will be possible in combination with blinking, after all!):

https://github.com/paradajz/OpenDeck/wiki/LED-control

paradajz, 2021-05-17:
Finally working on this for real. Will be part of next release. Software PWM method will be used so that brightness control is possible on any board and any pin (be it through shift register, native output or in matrix).

paradajz, 2021-05-18:
Resolved with https://github.com/paradajz/OpenDeck/commit/0c98723ceb36c5a99ae9f943d4ca2e3f0a741373

closed 2021-05-18 by paradajz