shanteacontrols/OpenDeck #22: Arduino uno and leonardo firmware error

4dvn, 2020-01-30:
I upload hex files with 2 boards but when open software can't find the board :(

paradajz, 2020-01-30:
You're not being very informative. How are you flashing the boards? Via flash.sh script? Is the output from the upload process fine? Does your PC recognize the newly connected MIDI device? Are you opening the web configurator from Google Chrome? Chrome is the only supported browser, although I supposed it works in Opera as well.

4dvn, 2020-01-30:
> You're not being very informative. How are you flashing the boards? Via flash.sh script? Is the output from the upload process fine? Does your PC recognize the newly connected MIDI device? Are you [...]

I flash using the old method of flip software on arduino and qmk on arduino leonardo

paradajz, 2020-01-30:
I don't know what's flip software and QMK. If you're flashing using external tools, you need to set the fuses manually, which is why flash.sh script is the recommended way. You can check the fuse settings here:

https://github.com/paradajz/OpenDeck/blob/master/src/Defines.mk#L33

closed 2020-02-13 by paradajz