shanteacontrols/OpenDeck #23: Short question about updates for the official board

amo13, 2020-02-27:
I just stumbled upon the releases section and I am not sure if the official openDeck boards still receive firmware updates, because of the disclaimer claiming incompatibility with official boards since a lot of releases ago.
I would like to buy an official board and build a controller, but I would also like to be able to update the boards when new firmwares are released. The disclaimer on every new release sounds to me as if the official board is not supported any further. Is this correct?

paradajz, 2020-02-27:
The current firmware is incompatible only with old versions of official board which had Mass Storage bootloader installed. Boards with that bootloader aren't shipped for a long time now, so all firmware releases are compatible with the board I'm currently shipping.

However, now that you mention it, I have just released new version of the firmware. This firmware will most likely be last firmware with new features for the current official board since it barely fits inside MCU anymore. For that reason, next board will be based on STM32F4 MCU. Current official board will still receive bug fixes if any are found, though. New board will come sometime later this year, most likely after summer.

closed 2020-02-27 by paradajz

amo13, 2020-02-27:
Awesome, thank you for your fast reply!

Nice to read about this year's further plans, too. Do you plan on making a board with more controls (like analog potentiometers) attachable to it or will it stay the same amount?

paradajz, 2020-02-27:
I've just published the new blog post which answers your question:
https://shanteacontrols.com/2020/02/27/opendeck-on-arm/

amo13, 2020-02-27:
Amazing news! You're killing it =)
The article mentions a same amount of inputs, but then lists 64 analog components, whereas the current opendeck board seems to work with only 32 analog components. Did I miss something?

paradajz, 2020-02-27:
You've read the part for OpenDeck Mega board which will also come later this year. To summarize, two new boards are coming:

1) One with the same amount of I/O and with same dimensions as the current official OpenDeck board, with additional I2C, SPI and UART connectors
2) OpenDeck Mega with support for 128 buttons, 128 LEDs, 64 analog components and dual DIN MIDI connectors

amo13, 2020-02-27:
Awesome, thanks for helping me understand.
Looking forward to all this already!