shanteacontrols/OpenDeck #24: LED Local - CC/S not working after update

Momecmusic, 2020-03-16:
Hi Igor,

I have a v1.1 board and had some issues accessing the configurator, so I just updated the board to v3.2.1

I have a very minor issue now that I can not get the led's to activate with "Local - CC/S". I have some momentary buttons with build in led's set up as "Latching" type, Midi message type "CC/0 off".

Before I updated it was all working fine, the corresponding led's where lit when activating the button and the correct midi CC value were sent. Pushing it again set the midi CC value to 0 and turned the led off. (Nice of you to implement this btw, it was a feature I asked for).

Now the led's only blink very shortly and only once in a while.

Hope this makes sense to you and that you can help me. 

Also: 1.1 boards can not be updated beyond v3.2.1 as I understand it?

Thanks,

Morten

paradajz, 2020-03-16:
Do you maybe have a video of the behavior you're noticing?

Yes, boards before v1.2.1 can't be updated anymore unless they're flashed directly with the new firmware or if they're flashed with HID bootloader.

paradajz, 2020-03-16:
Also, please try to configure it with the configurator v3.2.1:
https://github.com/paradajz/OpenDeck/releases/tag/v3.2.1

Momecmusic, 2020-03-17:
Hi Igor, 

thank you for getting back to me. Here is a video of the bug:

https://www.youtube.com/watch?v=_seORRyLUi0

Notice that the corresponding pad on the drum machine is lighting up as I push the button, normally the led on the opendeck controller would light up with it. It is a bit difficult to see, but the led briefly flashes very sporadicly.

I have the same behavior with the other led's.

And yes, I configured it with the v3.2.1 configurator :)

Thanks!

paradajz, 2020-03-17:
Hm. Tried factory reset first? Have you also tried configuring only that specific LED without touching others after factory reset?

paradajz, 2020-04-17:
Pretty sure this is the source of the issue: https://github.com/paradajz/midi-lib/commit/4d08a8e7e27195b69c4d1f28094e1cb294a0bd93

Fixed now on master, will backport it for official OpenDeck board.

closed 2020-04-17 by paradajz

forsakenrecordingstudios, 2020-04-26:
This is the same issue i am having. what was the fix?

paradajz, 2020-04-26:
The fix for this particular issue is to update the firmware to the latest version:
https://github.com/paradajz/OpenDeck/releases/tag/v4.1.0
https://github.com/paradajz/OpenDeck/wiki/Firmware-update#every-other-opendeck-board

forsakenrecordingstudios, 2020-04-26:
The web configuration says i am up to date

paradajz, 2020-04-26:
The latest version is 4.1.0

-------- Original Message --------

forsakenrecordingstudios, 2020-04-26:
Okay so can i update it through the web configuration or do i need to do it manually?

paradajz, 2020-04-26:
Web configurator doesn't support firmware update at the moment, you need to do it manually.

forsakenrecordingstudios, 2020-04-26:
so all i need to do is delete the firmware folder and paste in the new? thats it?

paradajz, 2020-04-26:
No. Please read that wiki page carefully. You have HID bootloader on board.

forsakenrecordingstudios, 2020-04-26:
i am but have no idea what this is

Windows: Open hid_dfu_windows.bat in scripts directory

paradajz, 2020-04-26:
What is unclear? It is a script you need to open. There is a `scripts` directory in OpenDeck repository.

forsakenrecordingstudios, 2020-04-26:
the opendeck board is not listed in the choices only these.
2 - Arduino Mega
3 - Arduino Uno
4 - Arduino Leonardo
5 - Arduino Pro Micro
6 - Teensy++ 2.0

paradajz, 2020-04-26:
Ahh damn, I thought GitHub handled releases differently. Please go here:
https://github.com/paradajz/OpenDeck/tree/v4

Click on "Clone or download", then Download zip.

forsakenrecordingstudios, 2020-04-26:
okay so when i select the Official OpenDeck board it just closes

paradajz, 2020-04-26:
Is your board in bootloader mode?

forsakenrecordingstudios, 2020-04-26:
yes all 3 lights are lit up

forsakenrecordingstudios, 2020-04-26:
i just type 1 and press enter correct?

paradajz, 2020-04-26:
Right. However there was a bug in the script. Please download the release again:
https://github.com/paradajz/OpenDeck/tree/v4

forsakenrecordingstudios, 2020-04-26:
okay have it all updated but LED's are still not activating with the buttons.

paradajz, 2020-04-26:
Okay, at least you're on latest firmware.

1) Which LED control mode are you using?
2) Are you controlling the LEDs via USB or DIN?
3) What MIDI messages are you sending to the board?

forsakenrecordingstudios, 2020-04-26:
BUTTON 64                    LED 1
Latching                         Activation ID 63
CC/0 off                         Activation Value 127
Midi Chan 1                   Midi Chan 1
Midi ID 63                     RGB no
On Velocity 127            Control Type Local - CC/S
                                      Blinking no
                                      State Test off

forsakenrecordingstudios, 2020-04-26:
BUTTON 64 
Latching 
CC/0 off 
Midi Chan 1 
Midi ID 63 
On Velocity 127 

LED 1
Activ ID 63
Activ Value 127
Midi Chan 1
RGB no
Control Type Local - CC/S
Blinking no
Sate Test off

forsakenrecordingstudios, 2020-04-26:
and it is USB

paradajz, 2020-04-27:
Thanks. It was another bug unfortunately.
https://github.com/paradajz/OpenDeck/commit/7527d16be456a8806b3add8cd7662cf5ab45dc51

New release is now out: v4.1.1
https://github.com/paradajz/OpenDeck/releases/tag/v4.1.1

Download "Source code (zip)" and update the firmware same way as before.

forsakenrecordingstudios, 2020-04-27:
Okay i will give it another try