shanteacontrols/OpenDeck #25: Mega R3 16u2

forsakenrecordingstudios, 2020-04-06:
Hello,

    I have attempted to get a daw plugin controller sending MIDI CC messages working for weeks now. Git everything working except encoders with another library and Ardiuno IDE. I ordered a opendeck board weeks ago but still waiting on it to arrive. I have flashes the mega so it is USB native. What else do i need to do to program this board with OpenDeck?

Additionally: i have 3 expanders i would like to use so i can set up 23 encoders on the Mega board. Is this doable with opendeck software?

paradajz, 2020-04-06:
The documentation for flashing the Arduino boards is here:

https://github.com/paradajz/OpenDeck/wiki/Flashing-the-OpenDeck-firmware

---

What kind of expanders?

-------- Original Message --------

forsakenrecordingstudios, 2020-04-06:
Yes i am just lost with the process. Does it have to be done that way in Windows or could i use flip? 

And the expanders are MCP23017

paradajz, 2020-04-06:
What is flip? I've tried my best to make the instructions as clear as possible. List of requirements is clearly listed. What exactly are you having issues with?

MCP23017 are not currently supported. 74HC165 is supported register for increasing the number of inputs and 74HC595 for outputs. 4067/4051 are supported multiplexers for analog inputs.

forsakenrecordingstudios, 2020-04-06:
Flip is a software for flashing devices. It was recommended by moco lufa when i flashed the board to make it usb native. It was only like 3 steps.

The problem im having is i just get a ton of errors

paradajz, 2020-04-06:
"Ton of errors" means nothing me without some logs

closed 2020-04-08 by paradajz