shanteacontrols/OpenDeck #26: MIDI IN trouble

Kubbik1, 2020-04-13:
Hi Igor,

your project is amazing and I already made a few midi controllers thanks to it.
Now I’m using Arduino Mega2560 and I have an issue with MIDI IN through DIN connector. In Web UI I activated MIDI DIN a connected pin 18 and 19 as described in your Wiki. MIDI OUT works fine but MIDI IN doesn’t react at all.
My goal is to control LEDs connected to Arduino over MIDI DIN from another device. Specifically I’d like to connect OpenDeck to another similar project - [PedalinoMini](https://github.com/alf45tar/PedalinoMini).
I’m running out of ideas what I’m doing wrong and will be grateful for any advice you can give me.

Thanks

Jim

paradajz, 2020-04-13:
Glad it (mostly) works out for you. Which firmware version are you running?

Kubbik1, 2020-04-13:
I have latest firmware v4.0.9, but the issue was the same with the lower version.

paradajz, 2020-04-13:
You're right, DIN MIDI in was broken. :( I've just pushed the fixes. Until I make a new release (have some more things planned for next one) you can compile the firmware yourself and then flash it like:

    make TARGETNAME=fw_mega2560
    make flash PORT=<your_port>

Kubbik1, 2020-04-13:
Thank you for quick response. I will try it.

Kubbik1, 2020-04-17:
Hi Igor, 
after I flash new fixed update, I still have the same problem. Maybe I set something wrong. When I enable DIN, do I have to enable Midi merge or anything else? I want to connect Opendeck to my main controller through din. Buttons on OpenDeck (powered with batteries) will send message through MIDI OUT to my main controller. And from PC through main controller I want to control LEDs on OpenDeck. Am I doing something wrong?
[image]

paradajz, 2020-04-17:
DId you flash the firmware by building the firmware and then running make flash or via flashing scripts?

Kubbik1, 2020-04-17:
I cloned new fixed OpenDeck.git, then in wsl Ubuntu built new firmware and then run make flash.

paradajz, 2020-04-17:
Ouch!! Turned out I had another bug in the MIDI library itself. Should be fine now, I've just tested it (I actually just tested DIN MIDI last time). Can you clone the entire repo again and make/flash it?

Here's the patch:
https://github.com/paradajz/midi-lib/commit/4d08a8e7e27195b69c4d1f28094e1cb294a0bd93

Kubbik1, 2020-04-17:
Still not working.

paradajz, 2020-04-17:
What messages are you sending to the OpenDeck board? Did you configure the LEDs

Kubbik1, 2020-04-17:
When I connect OpenDeck with USB cable and send midi from Ableton, Leds are working normal and tx/rx leds on mega are blinking. Connection through MIDI IN din is without reaction. I will try connect other midi device tomorrow. Thank you for your support.

paradajz, 2020-04-17:
Please send me the output that you're sending into OpenDeck board through DIN MIDI.

Kubbik1, 2020-04-17:
I am sorry, but I have no idea, how to do that.

paradajz, 2020-04-17:
Well, I need to know what exactly are you feeding into OpenDeck board through DIN MIDI. Could be that the data is somehow different sent through DIN MIDI from the USB messages. Sent message to the board must match channel, value and index in order to turn the specific LED on.

What operating system are you using?

paradajz, 2020-04-17:
Also, DIN MIDI merge should be off!

Kubbik1, 2020-04-17:
I have win 10. For example in ableton I make midi clip with note C3 (60 midi ID I thing) 127 velocity and send it through channel 1 to my device and Led1 connnected to pin 17 on Mega. When I directly connect Opendeck through usb, everything works great. My goal is send same midi from Ableton to RTP MIDI and wirelessly to device PedalinoMini (Esp32) and than send through din to OpenDeck. I checked PedaliniMini is sending message, but my mega isn't reading it.

paradajz, 2020-04-17:
Are you sending the signals directly pin-to-pin or actually through DIN MIDI connector?

Kubbik1, 2020-04-17:
I tried both.

paradajz, 2020-04-17:
Pin-to-pin wouldn't work since Pedalino is based on ESP32 which is 3.3V (or 3?) Anyways Arduino is 5V so you should definitely use DIN MIDI, with resistors and optocoupler.

What I've tried is to simply flash two Arduino Megas, enabled DIN MIDI on both of them, connected button to first one, connected TX from the first to the second (RX) and connected LED on second Mega. Pressing the button on first Mega lights up the LED on second Mega.

Kubbik1, 2020-04-17:
Good idea. I'll try it too. And then I try connect my keyboard.

Kubbik1, 2020-04-19:
You are awsome. Everything works now. Thank you.
[https://www.youtube.com/watch?v=Xobi0TV-1hA](url)

closed 2020-04-19 by Kubbik1

paradajz, 2020-04-19:
Very glad to hear that. :)