shanteacontrols/OpenDeck #27: Button with multiple, configurable OUTPUT (VAL).

gandinin, 2020-04-14:
Hello Igor,
in using OpenDeck to control VST Synthesizer, I don't simply need potentiometers and ON-OFF buttons but I often face the need to control "MULTI STATE" switches with more than just two states (rotary switches). 

An example are buttons (or rotary switches) used to choose Oscillator's waves forms, that often permit to choose between triangle, sawtooth, sine wave.
These types of controls are now performed trough sliders but could be, more profitably, managed by buttons with multiple output "VAL". An example, is the LFO slider on ARTURIA CS-80 VST that reacts to the position of a sliders to change between its three states:

MONO: val= 0-83
TRIG:   val= 43-84
FREE:  val=85-127

Considering the high number of button available in the board, but the relatively small number of potentiometer, I would love to avoid using potentiometers for these type of controls, but just address some of the several buttons and configure them so that they sends out not just extreme "VAL" (0 or 127), but also "n" intermediate (configurable) VALs. 
I’m not an expert in MIDI and I don’t know if this is something reasonable, but I think it can be useful for the users, that “STEP sliders” like the Yamaha CS80 but also the rotary switches of the Jupiter 8-v, or the three state button of the Farfisa, or the three position sliders of the Oberheim Sem V are controlled by "**MULTI VAL BUTTON**" (see attached pictures) .

Hope this help…

Best regards and thanks for the nice project.

Nicola

[image]
[image]
[image]

paradajz, 2020-05-31:
This is now implemented.

There are 4 new options for button configuration:
1) Multi value IncReset Note
2) Multi value IncDec Note
3) Multi value IncReset CC
4) Multi value IncDec CC

When the button is configured in any of these modes, "On velocity/control value" is treated like "step" value between two values. For example, if first option is selected, and "On Velocity" is set to 31, this is the output from the button (single message per button press, newest to oldest even):

```
21:7:58-Note On 48 vel 127 CH 1
21:7:58-Note On 48 vel 124 CH 1
21:7:58-Note On 48 vel 93 CH 1
21:7:57-Note On 48 vel 62 CH 1
21:7:57-Note On 48 vel 31 CH 1
21:7:56-Note On 48 vel 127 CH 1
21:7:56-Note On 48 vel 124 CH 1
21:7:55-Note On 48 vel 93 CH 1
21:7:54-Note On 48 vel 62 CH 1
21:7:54-Note On 48 vel 31 CH 1
```

Once the value reached 127, next press will start from configured velocity (step) value.

Second option is different - once the button reaches max value, next press will decrement the value until 0 is reached. This is more akin to how the actual multi-position switches behave:

```
21:13:56-Note On 48 vel 127 CH 1
21:13:55-Note On 48 vel 124 CH 1
21:13:55-Note On 48 vel 93 CH 1
21:13:55-Note On 48 vel 62 CH 1
21:13:54-Note On 48 vel 31 CH 1
21:13:54-Note On 48 vel 0 CH 1
21:13:54-Note On 48 vel 3 CH 1
21:13:53-Note On 48 vel 34 CH 1
21:13:53-Note On 48 vel 65 CH 1
21:13:53-Note On 48 vel 96 CH 1
21:13:52-Note On 48 vel 127 CH 1
21:13:52-Note On 48 vel 124 CH 1
21:13:51-Note On 48 vel 93 CH 1
21:13:51-Note On 48 vel 62 CH 1
21:13:50-Note On 48 vel 31 CH 1
```

This option isn't available for current OpenDeck board - it will be available on new ones later this year. For now the option works on any other supported board (Arduino/Teensy).

closed 2020-05-31 by paradajz

gandinin, 2020-05-31:
Wow!!!! Love that!
Thanks a lot for your effort. I really appreciate it.
I hope that this new feature will be appreciated by your customers as well. I think it's really a good one.
By the way, using your device, I think it's really an interesting piece of hardware.

Nicola