shanteacontrols/OpenDeck #28: Unable to configure SSD1306 connected to Mega2560

IrregularShed, 2020-05-01:
Hi!

I'm running OpenDeck 4.1.1 on a Mega2560 (a DIYMore Mega2560 Strong - very easy to hook up, so I recommend). I found some spare SSD1306 OLEDs in a parts box and thought I'd try and connecting one to the board, but the configuration stage is failing.

Going to the Display tab, both controller and resolution show 'Invalid':
[image]

I set them accordingly (note, here I've not enabled it - that makes no difference):
[image]

Now, when I try to go to another tab, the details are blank:
[image]

The DevTools console has an angular error showing - `Possibly unhandled rejection: {"errNumber":7}`.

Returning to the Display tab, that's completely empty now as well:
[image]

Clicking on the ℹ️ button at the bottom then throws the configurator back to the 'Searching for OpenDeck board' screen, but it's much darker than normal because a modal has tried to open, and the configurator is completely unresponsive. It also seems to crash the code on the board, as refreshing the configurator sees it unable to progress past the searching screen. The board needs to be unplugged and plugged back in before being able to use the configurator again.

This happens whether or not the OLED is actually attached, and as soon as the resolution is modified (even if set back to Invalid). Tested on Chrome and the standalone configurator on Windows.

paradajz, 2020-05-01:
I'll take a look on it as soon as I wrap up MIDI bootloader thing.

paradajz, 2020-05-02:
Should be resolved now, can you verify? Changes are on master currenty, you will need to build the firmware:

```
make TARGETNAME=mega2560
make flash PORT=some_port
```

IrregularShed, 2020-05-02:
I won't be able to until Monday - I'll let you know then!

IrregularShed, 2020-05-04:
OK, I built and flashed the master branch on my Linux box, which was successful. The two displays I tried are both 128x64 (but the boards are completely different) - neither of them would display anything. After trying to get that set up, the configurator stopped connecting to the board at all (using the github.io hosted version of it - didn't have time to try a local version). In the end I had to reflash the 4.1.2 firmware to be able to get the configurator to connect to anything, and when I did that the display tab was completely blank, and the log on the left side repeated Error a load of times).

I made sure the OLEDs were working by running the Adafruit SSD1306 Arduino library example code (like I said, I found them in a box of spares, and didn't know their state). They worked fine, however I had to change the I2C address from what was set in the example (0x3D for Adafruit's 128x64 OLED to 0x3C).

I'll try and look into it more, however we've had a death in the family today and I'm not sure what happens next with the Covid-19 lockdown in place 😟

paradajz, 2020-05-05:
It's most likely an address issue then, I need to take a look at I2C driver since it can block the entire app currently (not good!) on wrong address. Works fine for the displays I'm using so haven't bothered to look into that yet, but I will. I have some generic I2C SSD1306 displays from eBay.
-----
So sorry for your loss! :( Hope you're doing okay.

IrregularShed, 2020-05-05:
Yeah, just about ok. Didn't come as a shock, but the timing is horrible. Anyway, trying to keep my mind on other things.

If your display has 0x78 on the silkscreen on the back, that's probably one of the same displays as I've got. As I understand it, 0x78 is an 8-bit address (`01111000`) and 0x3C is its 7-bit equivalent (`00111100`) but I'm pretty far from an expert there 😉 I'll triple check the wiring set up later on with a logic probe to see if I can figure out what's going on.

paradajz, 2020-05-24:
Update!

1) I've totally reworked I2C API - if anything is messed up it won't halt the MCU anymore. https://github.com/paradajz/OpenDeck/commit/3754b7d9c3738e42c3bfdcca91c57b5ceb6f3f18
2) I've added the ability to define I2C address for display - it's currently exposed in Web interface. Current options are 0x00 (invalid, default), 0x78 and 0x7A. https://github.com/paradajz/OpenDeck/commit/559b6280768146ff3ee9acf7123ef912998f1a93

Would be cool if you could try it out again some time.

IrregularShed, 2020-05-24:
>    halt the MCU anymore. 3754b7d
>    default), 0x78 and 0x7A. 559b628

Amazing 🙂 I'll get it installed on Tuesday and see how it goes (I won't be
in my work area until then, otherwise I'd be at it now!)

IrregularShed, 2020-05-26:
Still not working for me, sorry. I just built and flashed the 5.0.0 firmware (checked on the info page) and then tried to enable a display. Nothing seemed to be happening so I disconnected and reconnected. Now the configurator thinks I've got 247 rotary encoders (first part of screen shown), and the log says ERROR WISH a lot:
[image]

And the version has gone completely wrong, showing either `v0.247.undefined` or `v1.247.undefined`
[image]

247 keeps coming up!

I haven't changed the bootloader on the 16u2 since I had it working, before I started trying to get displays hooked up. I'm assuming nothing has changed in there that could cause this, but will try reflashing that when I get some time.

paradajz, 2020-05-26:
Weird, works fine on my end with multiple boards. Try reflashing both 16u2 and 2560.

paradajz, 2020-06-05:
Sorry , I'll close this. Works fine here with multiple boards, and it works even on STM32F4 Discovery board. Don't really know what is wrong in your setup.

closed 2020-06-05 by paradajz

IrregularShed, 2020-06-05:
No problem. I've got WAY too much going on in life right this second to look into it, thanks for everything 👍