shanteacontrols/OpenDeck #29: Error when trying to flash pre-built binary (Pro Micro)

mungewell, 2020-05-20:
I have clone the repo, but I wish to flash pre-built rather than build myself. I think i am running the right commands but I get the following error
```
$ sudo ./scripts/flash.sh 
[sudo] password for simon: 
Please select serial port on which ArduinoISP is connected:
     1	ttyACM0
1
Please select board you want to flash and then press enter:
     1	arrival
     2	bergamot
     3	dubfocus
     4	jamiel
     5	mega16u2
     6	mega2560
     7	mega2560_6mux
     8	mega2560_jose
     9	opendeck
    10	teensy2pp
    11	uno
    12	uno16u2
Board number: 5
Please wait...
cat: ./build/DEBUG: No such file or directory
cat: ./build/BOOT: No such file or directory
Makefile:13: *** Target doesn't exist.  Stop.
```

mungewell, 2020-05-20:
I was able to flash the Pro Micro by back-tracking to
`commit 96dd93f738a39cb64ebbfec7e945be38df7e0e41 (HEAD, tag: v4.0.0)`

paradajz, 2020-05-20:
Fixed with https://github.com/paradajz/OpenDeck/commit/e82723886736cae89cf3cb7cad3b3bfd463947ac

Can you try again?

mungewell, 2020-05-20:
Late last night I realized that you had removed support for the ProMicro (not related to this bug BTW), with
```
commit cccddc142dd0c43b2dbc78ccbaea08ef20b411dc
Author: paradajz <2544094+paradajz@users.noreply.github.com>
Date:   Sun Apr 19 10:00:22 2020 +0200

    drop support for bootloader on arduino leonardo/promicro
    
    it won't fit anymore together with flash
```

If I grabbed the 'ProMicro.hex' from the previous rev and then used that together with HEAD I was able to flash the board with the fixed script.
```
git checkout 6c486272491c311df9cfefcfc67ed4d741951d4d
cp bin/compiled/merged/avr/atmega32u4/promicro.hex .
git checkout master
cp promicro.hex bin/compiled/merged/avr/atmega32u4/promicro.hex
sudo scripts/flash.sh
```

So, I consider this a success/fix, thanks.
[image]

paradajz, 2020-05-20:
It's not the ProMicro support that was removed - support for bootloader on ProMicro and Leonardo was removed. Glad it works though, and I'll also correct the naming in the UI.

closed 2020-05-20 by paradajz