shanteacontrols/OpenDeck #32: Fail to flash arduino2560

orzel, 2020-12-22:
Hello.

I've been trying to flash the arduino 2560 for two days. The main problem is that the board always report as "OpenDeck DFU", and i can't understand why. As a result the 'configurator' only allows firmware updates.

What I did:
It kinda worked very well. Everything went fine: I used a fresh git checkout/clone. From there I used the ./scripts/flash.sh to burn both the mega16u2 and the mega2560. Flashing did not report any error.

I tried connecting the board just after the mega16u2: it 'enumerated' as "OpenDeck", without the "DFU". But of course, it was not working as the main mcu was not yet flashed.
Once the mcu was flashed, it reported as "OpenDeck DFU".

What i tried to do to fix the problem:
* restart it all over: no success
* does it the other way around (flash 2560 then 16u2)
* flash firmware using a webbrowser on the configurator. It sees to work (progress bar + success popup message)
* i suspected a problem with eeprom content, so i : restored "arduino" configuration by reflashing both MCU, and used an arduino sketch to erase the whole eeprom (with '0'). Then re-install again.
* i tried using several previous versions from git
* wait very long after flashing, in reference to the note on https://github.com/paradajz/OpenDeck/wiki/Firmware-update#updating-firmware

Do you have any hint on what could have went wrong ?

AlpesMachines, 2020-12-23:
you need to flash the whole EEPROM with 0xFF if you want to restore.
i am particularly interested by this thread

Alpes Machines

alpesmachines.free.fr <http://alpesmachines.free.fr/>
alpesmachines@free.fr

paradajz, 2020-12-23:
Do you have some other board lying around? I've just tried the same procedure on my own Mega board and it works without issues.

paradajz, 2020-12-23:
The firmware on main MCU will boot bootloader instead of firmware if any of the following is true:

1) Hardware trigger is active. On Mega2560 board, this means that if MISO pin is tied to GND before power up, bootloader will be active (described here: https://github.com/paradajz/OpenDeck/wiki/Firmware-update#arduino-mega)
2) Software trigger is active: This is true if in Web UI you've rebooted to bootloader mode. On AVR boards such as Mega2560, this will write special value to special EEPROM memory location. On reboot, if bootloader reads this value, bootloader will be enabled. Specifically, last EEPROM location is used (on Mega this is 4095). If this value is 0x47, bootloader will activate. Any other value and application is loaded.
3) On startup, bootloader performs CRC check of firmware. If this validation fails, bootloader will be started instead of application. I suspect this is the part on your board that fails.
https://github.com/paradajz/OpenDeck/blob/master/src/board/common/bootloader/Bootloader.cpp#L66

orzel, 2020-12-23:
As said, i had tried erasing the eeprom. The script was writing '00' everywhere. Following AlpesMahines message, I also tried using 0xff, but it doesn't help neither.
Unfortunately, i dont have another board to check. Though i've used this one for lot of (arduino) stuff, and never had any problem with it.  I know about the hardware trigger, and even checked those pins to see if a short circuit was present. But nope.
Point 2 is probably not the reason, as i tried to zero the whole eeprom, and also to 0xff 

I guess there's no way to know which reason triggered the DFU mode ?

I can't see why the CRC check would fail, i tried several times to flash everythingi (both MCU). I would like to try to disable the check, but i can't compiler here. Most scripts don't work here on linux. I already had to patch the 'flash.py' script to make it work here. Other scripts have lot of problem. It seems those scripts were mostly written for windows WSL, not actual linux.

paradajz, 2020-12-23:
No, there isn't a way to tell which trigger activated bootloader.

> Most scripts don't work here on linux.
Huh?

> I already had to patch the 'flash.py' script to make it work here
Huh again? What did you change? It's also not a python script, it's a bash script, extension is .sh

Which avrdude version are you using? What programmer are you using? What distro are you on?

orzel, 2020-12-23:
Yes, sorry, bash script.
No big deal for the change, it's because I use the usbasp programmer, not avrisp,  so i had to remove the stuff about port (avrdude doesn't need a port for usbasp).

diff --git a/scripts/flash.sh b/scripts/flash.sh
```
index b2bda369..ae870acc 100755
--- a/scripts/flash.sh
+++ b/scripts/flash.sh
@@ -1,6 +1,7 @@
 #!/bin/bash
 
 run_dir="OpenDeck"
+find=find
 
 if [[ $(basename "$(pwd)") != "$run_dir"* ]]
 then
@@ -14,34 +15,6 @@ then
     exit 1
 fi
 
-if [ "$(uname)" == "Darwin" ]
-then
-    find="gfind"
-
-    if [[ "$(command -v gfind)" == "" ]]
-    then
-        echo "ERROR: GNU find not installed (gfind)"
-        exit 1
-    fi
-
-    port_list=$($find /dev -name "cu.*" -ls 2>/dev/null | sort | cut -d / -f3 | grep -v Bluetooth)
-elif [ "$(uname -s)" == "Linux" ]
-then
-    find="find"
-    port_list=$($find /dev/serial/by-id/ -type l -ls | grep -Eo '\btty\w+')
-fi
-
-if [[ "$port_list" == "" ]]
-then
-    echo "ERROR: No ports found. Please connect ArduinoISP before running the script."
-    exit 1
-fi
-
-echo "Please select serial port on which ArduinoISP is connected:"
-echo "$port_list" | cat -n
-read -r port
-port=$(echo "$port_list" | head -n "$port" | tail -n 1)
-
 echo "Please select board you want to flash and then press enter:"
 
 boards=$($find bin/compiled -type f -name "*.hex" -path "*merged/avr*" -printf '%f\n' | sort)
@@ -64,6 +37,6 @@ high_fuse=$(command < src/board/avr/variants/avr8/"$mcu"/fuses.txt grep ^high= |
 echo "Connect programmer to programming header on the board and then press enter."
 read -rn1
 
-avrdude -p "$mcu" -P /dev/"$port" -b 19200 -c avrisp -e -V -u -U lock:w:"$unlock_fuse":m -U efuse:w:"$ext_fuse":m -U hfuse:w:"$high_fuse":m -U lfuse:w:"$low_fuse":m
-avrdude -p "$mcu" -P /dev/"$port" -b 19200 -c avrisp -U flash:w:"$path"
-avrdude -p "$mcu" -P /dev/"$port" -b 19200 -c avrisp -V -u -U lock:w:"$lock_fuse":m
\ No newline at end of file
+avrdude -p "$mcu" -b 19200 -c usbasp -e -V -u -U lock:w:"$unlock_fuse":m -U efuse:w:"$ext_fuse":m -U hfuse:w:"$high_fuse":m -U lfuse:w:"$low_fuse":m
+avrdude -p "$mcu" -b 19200 -c usbasp -U flash:w:"$path"
+avrdude -p "$mcu" -b 19200 -c usbasp -V -u -U lock:w:"$lock_fuse":m

```

orzel, 2020-12-23:
The kind of error i have if trying to compile is:

orzel@berlioz tmp/OpenDeck/src% make target=mega2560
usage: yq [-h] [--yaml-output] [--yaml-roundtrip] [--width WIDTH]
          [--indentless-lists] [--in-place] [--version]
          [jq_filter] [files [files ...]]
yq: error: argument files: can't open 'bootloader.use': [Errno 2] No such file or directory: 'bootloader.use'
usage: yq [-h] [--yaml-output] [--yaml-roundtrip] [--width WIDTH]
          [--indentless-lists] [--in-place] [--version]
          [jq_filter] [files [files ...]]
yq: error: argument files: can't open 'arch': [Errno 2] No such file or directory: 'arch'
usage: yq [-h] [--yaml-output] [--yaml-roundtrip] [--width WIDTH]
          [--indentless-lists] [--in-place] [--version]
          [jq_filter] [files [files ...]]
yq: error: argument files: can't open 'mcu': [Errno 2] No such file or directory: 'mcu'
usage: yq [-h] [--yaml-output] [--yaml-roundtrip] [--width WIDTH]
          [--indentless-lists] [--in-place] [--version]
          [jq_filter] [files [files ...]]
yq: error: argument files: can't open 'mcuFamily': [Errno 2] No such file or directory: 'mcuFamily'
Defines.mk:85: *** MCU  not supported.  Arresto.

Or using the shell script:

orzel@berlioz clones/OpenDeck/src% ../scripts/build_targets.sh --type=fw_release
usage: yq [-h] [--yaml-output] [--yaml-roundtrip] [--width WIDTH]
          [--indentless-lists] [--in-place] [--version]
          [jq_filter] [files [files ...]]
yq: error: argument files: can't open 'release': [Errno 2] No such file or directory: 'release'
usage: yq [-h] [--yaml-output] [--yaml-roundtrip] [--width WIDTH]
          [--indentless-lists] [--in-place] [--version]
          [jq_filter] [files [files ...]]
yq: error: argument files: can't open 'test': [Errno 2] No such file or directory: 'test'

orzel, 2020-12-23:
> > Other scripts have lot of problem. It seems those scripts were mostly written for windows WSL, not actual linux.

Nope, sorry.
I had to change some bash code related to $($xxx), and I assumed it was because of a difference between bash in wsl and bash in linux. But actually it was my fault. Sorry again.

paradajz, 2020-12-23:
I'm sorry but you're using USBasp here instead of supported programmer which is Arduino as ISP. I had some troubles with USBasp in the past which is why I'm not using it anymore, nor it is mentioned anywhere in the documentation or the code. Please don't open issues for stuff I don't support.

closed 2020-12-23 by paradajz

orzel, 2020-12-23:
Oh. You're rude.