shanteacontrols/OpenDeck #33: Loss of 5-pin midi control in standalone with Arduino Mega R3 -based controller

mikelcoolj, 2021-01-13:
When connected to web configurator, my Arduino-based controller sends midi over 5-pin seamlessly, and will continue to do so until connection with computer is reset.  Accessing the web configurator seems to activate the device, after which it will then continue to function properly--even with web configurator closed--until usb is disconnected.  Accordingly, the device will not send midi over 5-pin or usb when operating in standalone, whether powered by 9v or usb.  I'm using firmware v5.2.0.  Thank you!

paradajz, 2021-01-13:
Fixed with https://github.com/paradajz/OpenDeck/commit/608f8e1348617a2a90baca9a467d5bcab332ab6c
New release is available here:
https://github.com/paradajz/OpenDeck/releases/tag/v5.3.0

mikelcoolj, 2021-01-14:
Working perfectly now, thanks so much!

closed 2021-01-14 by paradajz