shanteacontrols/OpenDeck #123: Blackpill flashing process completes, but device does not appear.

mattjoyce, 2022-01-07:
I have followed the very clear flashing guide for an STM32F401CCU6, on Windows 10, using v6.2.4  blackpill401.bin
The experience goes exactly as described in the guide, except, that after the final message "Transitioning to dfuMANIFEST state", no new device appears in the Device Manager\Sounds   After flashing I hear the USB device disconnection sound and 'STM32 BOOTLOADER' in the 'libusbK USB Devices' section disappears, but I do not hear a connection sound.  If I unplug or reconnect the device, I do not hear any sound notifications and devices do not change.

I have also tried v6.2.3 with the same result.

Found DFU: [0483:df11] ver=2200, devnum=5, cfg=1, intf=0, alt=3, name="@Device Feature/0xFFFF0000/01*004 e", serial="339D36483335"
Found DFU: [0483:df11] ver=2200, devnum=5, cfg=1, intf=0, alt=2, name="@OTP Memory /0x1FFF7800/01*512 e,01*016 e", serial="339D36483335"
Found DFU: [0483:df11] ver=2200, devnum=5, cfg=1, intf=0, alt=1, name="@Option Bytes  /0x1FFFC000/01*016 e", serial="339D36483335"
Found DFU: [0483:df11] ver=2200, devnum=5, cfg=1, intf=0, alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,01*128Kg", serial="339D36483335"

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 239192
Download        [=========================] 100%       239192 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state

I have tried two different Blackpill devices, same model.

Any ideas?

paradajz, 2022-01-07:
Weird. I have the same model hooked up on my hardware test rig and everything works just fine. Would it be possible for you to try the same thing in Ubuntu?

paradajz, 2022-01-07:
Oh I figured it. You've hit the joy of having almost-the-same-but-then-not-quite variant of the STM32F4 MCU.

The board I have features F401**CE**. You have F401**CC**. The difference is the amount of RAM and flash - yours has 256kB of flash as opposed to 512kB on CE variant, so flash pages aren't correctly generated. The board goes to hard fault mode straight after flashing most likely. I'll add a new variant later today so you can flash that.

mattjoyce, 2022-01-07:
Ah, thank you that would be ace.
I was about to pull an old laptop out to install Ubuntu.  However out of curiosity, I tried it on an Mac Air M1, and got the same result.  Slightly later version of dfu-util, which does provide hints that all is not well.

Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 239192
Erase           [=========================] 100%       239192 bytes
Erase    done.
Download        [=========================] 100%       239192 bytes
Download done.
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state

I had to install homebrew to get dfu-util, using this command.
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"

paradajz, 2022-01-08:
`dfu-util` is mentioned in packages to install. ;)

I've attached test binary for F401CC. Can you try it? I don't have that board so I can't test it. Note that only 1 preset is supported on that board.

[blackpill401cc.zip](https://github.com/shanteacontrols/OpenDeck/files/7833756/blackpill401cc.zip)

mattjoyce, 2022-01-08:
> dfu-util is mentioned in packages to install. ;)
Yes, I was just leaving the command for other M1 users to install homebrew :)

OK, well the new binary seems to work a treat.  Flashed, and restarted, appears as in Device Manager as described in the guide.
The offline OpenDeck UI detects the device!

Thank you, what wonderful support, and what an interesting project.  I spent many weeks trying to write my own with Arduino, I'm looking forward to converting the thing I made to work with OpenDeck.

I will get something attached to properly test later today.

paradajz, 2022-01-08:
Awesome. The CC variant will be part of the new release in a day or so, so you can grab future updates from GitHub or update through web tool.

mattjoyce, 2022-01-09:
Yep, I tested a simple analog pot and mapped it through to VCV Rack, works very well.  Thank you!!

paradajz, 2022-01-09:
Solved with:
https://github.com/shanteacontrols/OpenDeck/commit/c85396f1bd41f26abb84ec2b7a5f16964b6d629d
https://github.com/shanteacontrols/OpenDeck/commit/84b673b20a10bbf4b87676b4de15062ba15d840f

closed 2022-01-09 by paradajz

paradajz, 2022-01-11:
Added to the new release:
https://github.com/shanteacontrols/OpenDeck/releases/tag/v6.3.0