shanteacontrols/OpenDeck #137: STM32F4 Discovery

Wapata, 2022-02-02:
Hi !
I did put the discovery.bin file into my (new) STM32F4 Discovery, seen the blinking LED, unplugged my MINI-USB, linked PA9 to 5V, plugged the MICRO-USB and then... nothing. No device into windows, into my midi software or into the configuration page.

DETAILS.TXT say that :
Version: 0221
Build:   Nov 19 2015 15:23:07

Do you have an idea of what could be wrong ? Thanks !

Wapata, 2022-02-02:
The midi IN/OUT din are not mandatory I think. It's should appear as a midi native device and AFTER i can activate midi-din stuff...right ?

paradajz, 2022-02-02:
Which F4 Discovery exactly do you have?

Wapata, 2022-02-02:
Thanks for the answer.

Front of paper
 part number STM32F4DISCOVERY from www.st.com/stm32f4-discovery 
 back of the paper
32F407GDISEM/ 01-0  order code STM32F407G-DISC1

On pcb
STM32F407G-DISC1
MB997D
Label in back of pcb
MB997 D-01
2171u0476

...
bouth from Farnel in 2017. Don't know if you need something else ?
Oh... USB cable checked and okay (two mini and two micro)

paradajz, 2022-02-02:
I've just done exactly what the guide says - copied the bin file, waited few seconds, the board rebooted and new MIDI device appeared. Did you try with both mini and micro USB cables connected?

Wapata, 2022-02-02:
Never both at the same time. Should I ?

paradajz, 2022-02-02:
Yes, please do.

Wapata, 2022-02-02:
Nope. flash and then nothing.
Jumpers on JP1, DISCOVERY (both), JP2 and JP3 (so on everything)

paradajz, 2022-02-02:
Don't know then, really. I have two of those boards here and both work fine. Which OS are you using?

Wapata, 2022-02-02:
Yes it's windows. I will try with Linux at the end of the week then.
(All my custom midi devices work great on my computer).
I don't find what 3 or 4 flash mean by the way ..
Thanks for your help !

Wapata, 2022-02-05:
In linux it failed too BUT a file is disposed into the device.
FAIL.TXT say:
The interface firmware FAILED to initialize the target MCU

Wapata, 2022-02-05:
Okay... trying an other pair of cables and it work :'-(
... I have to found the instructable that show ow to make a USB cable health check.

I am SO sorry about that. I have to put labels on theses "bad data / power ok" cable :-(

closed 2022-02-05 by Wapata

paradajz, 2022-02-06:
This might be for you:
https://www.tindie.com/products/nerfhammer/is-it-me-or-is-it-usb/