shanteacontrols/OpenDeck #138: STM32F4 Discovery don't AutoSave

Wapata, 2022-02-05:
Hi !
So I have a STM32F4 Discovery board with button 0 and pot 0 connected and working.
If I change the note number (or the CC number) in the web or desktop or web-android UI, it work great but if i unplug or reboot... it goes back to "factory" settings.

Did i miss a big save button anywhere ?

Thanks

paradajz, 2022-02-06:
There is no save button. The settings are persistent across reboots. I can't reproduce this. Maybe try factory reset through UI and then try again?

Wapata, 2022-02-06:
Thanks for the answer.
Already try factory reset (and some other USB cable ...) But nothing.
Did backup and restore too but as soon as I unplug , everything is back to factory.

Maybe I should put away the discovery board ?..
What is the next cheep board I can play with ? I'm looking to use 6 pots and 6 buttons for the "classics" CC messages (and each presets will use his own midi chanel).
Still hesitate between a full screen-controler or a full physical one.

paradajz, 2022-02-06:
There is something wrong with the binary attached to GitHub release. Investigating. Can you try the attached one?
[discovery.zip](https://github.com/shanteacontrols/OpenDeck/files/8010214/discovery.zip)

Wapata, 2022-02-06:
It's working perfectly !
Thanks a lot for the helping hand. I was worried of the few informations i was able to give to you... but you did the magic !

paradajz, 2022-02-06:
Well I haven't really... Still need to figure out what is wrong with the binaries attached to release. Ugh. Glad it's working out. I'll leave this opened until the binaries on GitHub are fixed.

paradajz, 2022-02-06:
Released the new version, can't reproduce the original issue anymore so I guess it's fine:
https://github.com/shanteacontrols/OpenDeck/releases/tag/v6.3.2

closed 2022-02-06 by paradajz

Wapata, 2022-02-06:
Updated via UI and it work here too !

Bravo and keep making great stuff !
Is there a place for me to put the first picture of my prototype ? It's ugly but it's working so great that it take more time to heat up the solder iron that programming....

paradajz, 2022-02-06:
You can share it in Discussions/show and tell:
https://github.com/shanteacontrols/OpenDeck/discussions/categories/show-and-tell