shanteacontrols/OpenDeck #166: Unable to select display I2C adress in web configurator

basst22778, 2022-08-22:
i am trying to chance the i2c adress in the web configurator as discribed in your comment.
https://github.com/shanteacontrols/OpenDeck/issues/28#issuecomment-633228800

There is no option in the interface. (bluepill, V7.3.2)

paradajz, 2022-08-22:
Can you post a screenshot?

basst22778, 2022-08-22:
(url
[image]
)

paradajz, 2022-08-22:
Ah, right. The option has been removed because the code tries to automatically determine the address of the display. Right now the two probed addresses are 0x3C and 0x3D.

basst22778, 2022-08-22:
[image]
OK - then i have the wrong display. 
its 0x78 in the specs.

paradajz, 2022-08-22:
0x78 is actually 0x3C and should work fine, however, I have run into some displays from eBay that simply do not work at all.

closed 2022-08-22 by paradajz