shanteacontrols/OpenDeck #176: RPI Pico W + 4067 multiplexer

walrusmustgo, 2022-11-08:
Hello! First of all, i'd like to thank you for this wonderful project! It kickstarted my entry to the vast electronic world! ^^ 

With that being said, i'm a noob, just started fiddling with electronics. Now,i've got an issue with using multiplexers with the rpi pico w board. If i connect a single potentiometer to the ADC input of the Pico, everything is working correctly, but if i build a new firmware, with multiplexers enabled, i get really hectic behaviour once the pot gets under a certain threshold, as you can see here:

https://user-images.githubusercontent.com/57779757/200574837-562322bb-f68f-431c-b88c-73aa296cf710.mp4

My board is setup correctly ( as far as i'm aware ):

[image]

This is the build config:
[image]

Upon measuring voltages, the pot sends good values to the multiplexer, but on the z0 pin, the voltages are much smaller ( 3.3V on the channel pin, ~0.8V on the signal pin ). 

As i've said, i'm a noob, so i might be missing something fundamental/obvious! :) 

Thanks again for this beautiful piece of software, and your hard work! 

Regards, Marko.

paradajz, 2022-11-08:
Can you post your entire config file?

walrusmustgo, 2022-11-08:
`---
  mcu: "rp2040"
  usb: true
  uart:
    dinMIDI:
      pins:
        rx:
          port: "0"
          index: 1
        tx:
          port: "0"
          index: 0
    touchscreen:
      pins:
        rx:
          port: "0"
          index: 5
        tx:
          port: "0"
          index: 4
      components: 64
  i2c:
    display:
      pins:
        sda:
          port: "0"
          index: 16
        scl:
          port: "0"
          index: 17
  buttons:
    type: "native"
    extPullups: false
    pins:
    -
      port: "0"
      index: 2
    -
      port: "0"
      index: 3
    -
      port: "0"
      index: 6
    -
      port: "0"
      index: 7
    -
      port: "0"
      index: 8
    -
      port: "0"
      index: 9
    -
      port: "0"
      index: 10
    -
      port: "0"
      index: 11
    -
      port: "0"
      index: 12
    -
      port: "0"
      index: 13
  analog:
    type: "4067"
    inputVoltage: "3.3"
    multiplexers: 1
    pins:
      s0:
        port: "0"
        index: 18
      s1:
        port: "0"
        index: 19
      s2:
        port: "0"
        index: 20
      s3:
        port: "0"
        index: 21
      z0:
        port: "0"
        index: 26
  leds:
    external:
      type: "native"
      pins:
      -
        port: "0"
        index: 14
      -
        port: "0"
        index: 15
      -
        port: "0"
        index: 22
      -
        port: "0"
        index: 21
      -
        port: "0"
        index: 20
      -
        port: "0"
        index: 19
      -
        port: "0"
        index: 18`

Here it is! :)

paradajz, 2022-11-08:
You are sharing the pins between LEDs and multiplexers: pins 18, 19, 20 and 21 are defined both for LEDs and multiplexers. Remove the LED ones.

paradajz, 2022-11-08:
Did you run "make clean" after you've changed the config file?

walrusmustgo, 2022-11-08:
Tried it now, rebuilt, but the problem still persists :(

paradajz, 2022-11-08:
> Do i have to play around with ADC offset by any chance?

No. Running the 4067 mux on that same ADC pin on my test setup works fine. My board is standard Pico, though, not the W version, so I'm not sure. Double check your mux wiring, maybe there's a loose contact somewhere.

paradajz, 2022-11-08:
You should also tie any unconnected mux pins to GND - floating pins cause noise.

paradajz, 2022-11-08:
Please also verify that you are on latest commit on master. After that, to rule out issues with the mux, disconnect Sx and common/signal/Z pin from Pico, tie all Sx pins to GND, and then measure the voltage on the signal pin when the pot is turned all the way left and right (so the voltage readouts should be 0V and 3.3V, respectively).

walrusmustgo, 2022-11-08:
Alrighty!
1) The commit is latest, just tried cloning the repo again, did a git pull, no changes. Built again, the same issue.
2) I've tried 3 MUX boards with the method you've suggested, all mux boards display correct values (0V - ~3.3V)

I'm more and more starting to believe that it's the Pico W issue to be honest, I've tried building with different pins, some things have definitely changed for the Pico W board: 
[image]

I'll know more tonight, if it's a Pico W issue, than i'll see if i can pinpoint the problem, and i'll let you know if i've fixed it! And i'll let you know if regular Pico is working correctly :)