shanteacontrols/OpenDeck #222: I can't pack the project

juanhima1, 2023-08-01:
I have made few changes in the project, mainly translating texts. It works fine using "yarn dev".
I open Chrome and I can view my changes in "localhost:3002"
I'm trying to package the project with "yarn electron-packager . OpendeckConfigurator --overwrite --platform=win32 --arch=x64" but when I execute OpendeckConfigurator.exe and the window shows nothing. I used CTRL-SHIFT-I to show the problem. I saw that "index.html" couldn't find the file "main.ts". I solved this problem changing "<script type="module" src="/src/main.ts"></script>" for "<script type="module" src="./src/main.ts"></script>"
I have packaged the project again, and still is not working. This time. the error showed in the window of the OpenDescConfigurator refering to the "main.ts", is:
"Failed to load module script: Expected a JavaScript module script but the server responded with a MIME type of "video/vnd.dlna.mpeg-tts". Strict MIME type checking is enforced for module scripts per HTML spec."

I haven't made changes in files like "main.js", "main.ts", "tsconfig.json", "package.json".
To be sure, I downloaded [202306291358](https://github.com/shanteacontrols/OpenDeckUI/releases/tag/202306291358) again. I tried to package and the result was the same. I had to correct "index.html" and finally I get the same problem as I told above.

I'm sure that I'm doing something wrong but I have no idea what.

Any suggestions?
Thanks

paradajz, 2023-08-06:
wrong repository

closed 2023-08-06 by paradajz

juanhima1, 2023-08-07:
I don't know what you mean....
Where is the right repository?