shanteacontrols/OpenDeck #234: LED not turning off

devkon-at, 2023-09-20:
I am testing an OpenDeck v2 (Firmware 7.5.4)  with a LED Button (single color):
Configured via USB - Web Config tool

Button is at ID 63 (R8 C8)
LED is at ID 47 (R6 C8)

Button 63 Config:
Latching
Note
Midi Channel 1
Midi ID 63
Value 127

LED 47 Config:
color testing - off
Activation ID: 63
RGB Enable OFF
Control Type Local / Note (Single value)
Activation value 127
MIDI channel 1

We would expect the LED to turn on and off when pressing the button but the Voltage only changes minimally (4.5V to 4.1V). Do you have any idea what the issue could be?

Thanks

paradajz, 2023-09-22:
Which board revision exactly do you have?

flokoat, 2023-09-25:
We used:
OpenDeck board
Revision: 2.2.1
2022.12

We also tried LED ID 0 (R1 C1) with same results.

The only time we could observer the LED to go dark (around 1V on startup) with this general LED setting enabled:
Start-up animation ON
Enables or disables LED animation when the device is powered on.

Here is our current configuration:
[OpenDeckUI-Backup-2023-09-25-14-02.zip](https://github.com/shanteacontrols/OpenDeck/files/12716050/OpenDeckUI-Backup-2023-09-25-14-02.zip)
Rotary Encoder (ID 0) working
Analog 31 working
Button 63 working
LED 0 only jumping between 4.1V and 4.5V

paradajz, 2023-10-07:
Sorry for the late reply. I have just tried this and couldn't reproduce. Have you maybe tried with different LED?

flokoat, 2023-10-19:
It worked with a different LED, the butten with the built in LED also had an built in bridge rectifier which caused the problem.
Thank you for your help!

closed 2023-10-19 by paradajz