shanteacontrols/OpenDeck #239: OSC

pmjcontroller1, 2023-10-20:
Good morning.  He launched a proposal, I don't know if this is the right place or it should be in UI.  The proposal is if we could add the option that a field can be added in the UI to send OSC message.  It would be a great advance, we can collaborate with its development.

paradajz, 2023-10-20:
> He launched a proposal

Who? Collaborate with whom? OSC is a very loose specification, I'm not sure what kind of messages are you interested in.

pmjcontroller1, 2023-10-20:
Collaborating by making a donation can be.  It would be enough for the user in the configurator to be able to configure the input port, OSC output, and the destination IP.  On each button, fader and encoder enable a space for the user to write the necessary OSC message, this could be a starting point

⁣
paco Lop
Dpto comercial
+34 663390200
www pmjcontroller.es​