shanteacontrols/OpenDeck #241: Problems with china-encoders

gh0stless, 2023-10-22:
I have a problem with my new M-board and cheap China encoders from Ali-Express (with buttons). The encoders get "stuck" and you have to go back to the last one turned and turn it a little, then others will work too. Turning quickly will definitely cause it to jam.
Now it seems to me that these problems only occurred with 12 encoders (2 groups of 6), previously I had 8 encoders (2 groups of 4), and I don't remember any problems. Unfortunately I can no longer verify this.
Are these encoders from China just garbage?

paradajz, 2023-10-22:
Hm. Can you record a short video showing what's happening? Not quite sure I understood you.

gh0stless, 2023-10-22:
https://github.com/shanteacontrols/OpenDeck/assets/11757531/cd711655-0d95-4428-855d-8e012008d4ed

gh0stless, 2023-10-22:
I just switched off encoders 5 and 6 as well as 11 and 12, there are no problems here!

paradajz, 2023-10-22:
Okay. Pick one encoder with which you're having issues. Disconnect and disable all the other encoders, then try to spin it as fast as possible and see if there are any problems. If there are, try to figure out if the problem exists on specific pins only. If there aren't, gradually start enabling other encoders one by one. Try to find a pattern here.

gh0stless, 2023-10-22:
This is difficult because the encoders are soldered onto a PCB.
[image]

paradajz, 2023-10-22:
Okay, then just disable them.

gh0stless, 2023-10-23:
A single encoder always works, but even with 2 there are problems, I think the encoders are just junk, since it seems to work for all of you. By the way, I also had this problem with the old V.1 board.

paradajz, 2023-10-23:
Still not sure I understand. I need exact steps to be able to reproduce. What do you mean with 'single encoder always works'?

Is it:

1) All other encoders are disabled
2) All encoders are enabled but you are only ever moving a single encoder at the time?
3) All encoders are enabled and the problems start when you move encoders simultaneously?

Please be as specific as you can.

gh0stless, 2023-10-23:
Yes, 1, all other encoders are disabled.
I never move encoders simultaneoulsly.

gh0stless, 2023-10-27:
Hello Igor, would it be of any use to send you a few such encoders?

gh0stless, 2023-10-27:
[image]

paradajz, 2023-10-29:
Probably yes. But as you've already said, it's probably due to the encoders. I've never had issues with the ones I'm using from ALPS.

gh0stless, 2023-11-22:
I have now also bought these identical encoders from Amazon and installed them. The quality appears to be better, and “hangs” only rarely occur. But I still don't understand why the software doesn't discard false signals after a while.