shanteacontrols/OpenDeck #249: Problem with Analog Buttons

thomasf78, 2023-11-13:
Hi Igor
i have problems with Analog buttons on OpenDeck M (Board Version 2.2. // Firmware 7.6.0 - but also on older Firmware Versions).

I started the project without connected Buttons and activated the Analog inputs - as in wiki manual recommended.

If i connect the buttons (EC11 encoder buttons, and simple PCB Board Buttons) i have the same issues:
- Some Channels are ok 
- Some Channels deliver two different button number changes after each other

There is a pattern recognizeable.
e.g. 
Button 74 (Analog Input 10) - OK
Button 75 (Analog Input 11) - not OK --> I receive messages from both Channel 11 and Channel 10
Button 76 (Analog Input 12) - OK
Button 77 (Analog Input 13) - not OK --> I receive messages from both Channel 13 and Channel 12
Button 78 (Analog Input 14) - OK

At this point i havent connected Channel 0-9, but activated. There i will add some potentiometers.

Can you give me a hint what i am doing wrong.
Attached a Video with the Console Log

https://github.com/shanteacontrols/OpenDeck/assets/68353724/b1740709-4f57-47b2-a7c5-8f656702e2bb

Thanks so much
Best Thomas

paradajz, 2023-11-13:
I think you're confusing terms a bit here. Buttons aren't sending wrong channels, they are sending wrong MIDI indexes. Can you post how you have those buttons wired up?

thomasf78, 2023-11-13:
Hi Igor,
thanks for your fast reply.
Sorry about confusing you. 

I wired the buttons in the following manner:
Button 1 - Analog Port 10 (A:+ // B: Signal) --> AH2 
Button 2 - Analog Port 11 (A:+ // B: Signal)
Button 3 - Analog Port 12 (A:+ // B: Signal)
Button 4 - Analog Port 13 (A:+ // B: Signal)
Button 5 - Analog Port 14 (A:+ // B: Signal)
Button 6 - Analog Port 15 (A:+ // B: Signal) --> not shown in the video

My plan is to do this also on AH3 for six more buttons. Not realised yet.
Just for testing purpose i wired up a button on:
Button x - Analog Port 31 (A:+ // B: Signal) --> the same behaviour --> MIDI Index 31 and 30 sent.

I hope this is helpful.
Thanks and best 
Thomas

paradajz, 2023-11-13:
Did you use the resistors?

thomasf78, 2023-11-13:
hi igor. shame on me. i didn't. how i could forgot the important pulldown resistor. 
thats the reason why i got wrong signals. thanks a lot. i will change my layout. many thanks. i will inform you if it works.
best thomas

thomasf78, 2023-11-16:
Hi igor, thank you very much for your support. you can close the task.

@ everyone who is so stupid like i was, and has forgotten the pull down resistor -- here is the solution. :-)

closed 2023-11-21 by paradajz