shanteacontrols/OpenDeck #254: Mega 2560 not responding in web configurator

donnerbono, 2023-12-19:
Everything works perfectly from the build with no errors at all. Built on Ubuntu, used capacitor as Uno to Mega, flashed both microcontrollers, everything looks like it does in the wiki (and I know what it looks like to fail as didn't get pins right first time for example). Instructions are really thorough.

But web configurator doesn't recognise it at all. My Pi Pico is recognised but that's a much simpler build.

Aware this is my issue rather than a fault - but any pointers of anything to try as would be great to get this working as want more than 3 pots.

paradajz, 2023-12-20:
Not sure. Just tried with my Mega board with the latest firmware and everything works fine...

[image]

paradajz, 2023-12-20:
What operating system are you using? Is the board listed as MIDI device or not? Also, on initial flash, the board setup takes some time, half a minute maybe? Don't remember anymore. By setup, I mean the internal EEPROM memory gets formatted with initial data.

donnerbono, 2023-12-20:
Thanks for reply. Ubuntu 22.04 as recommended on site. I’ve got a blackpill I’m going to use instead and come back fresh to this. I would guess it’s a usb issue as everything ran perfectly.

closed 2023-12-30 by paradajz