shanteacontrols/OpenDeck #308: Cmake erroring due to missing build.ninja file

Pedalnetics, 2024-10-15:
After launching the docker image, if it finds ninja on the commandline the Makefile is adding -GNinja to the build args.  However, this is resulting it in trying to find a build.ninja file that does not exist:

`root@devlaptop:/opt/OpenDeck# ./scripts/dev.sh
Creating new container
Pulling opendeck ... done
Starting opendeck ... done
(.venv) ubuntu@opendeck ~/workspace/OpenDeck master
> make TARGET=mega2560
ninja: error: loading 'build.ninja': No such file or directory
make: *** [Makefile:37: all] Error 1
(.venv) ubuntu@opendeck ~/workspace/OpenDeck master
>`

After removing the offending lines in the Makefile that added -GNinja to build args, it then compiles fine.

Recommendation to either include a build.ninja file for Ninja to successfully work, or remove the following lines from the Makefile:
`ifneq (,$(shell which ninja))
    CMAKE_CONFIG_ARGS += -GNinja
endif
`

At least, that's what i had to do for this to work properly =)

paradajz, 2024-11-05:
I'm not sure how did you get into this state, should never happen and indeed, it never happened to me. Might be something specific with your setup.

paradajz, 2024-11-05:
Unable to reproduce.

closed 2024-11-05 by paradajz

Pedalnetics, 2024-11-05:
Not sure, I followed the Wiki to-the-T a few times, including deleting every cached docker image before retrying, and hit it multiple times.  If your make file is going to have a spot that adds GNinja, then there should be a build.ninja file; otherwise, that argument doesn't belong in there, at least that's my understanding

paradajz, 2024-11-05:
> If your make file is going to have a spot that adds GNinja, then there should be a build.ninja file

There shouln't. Cmake generates all ninja files first.

Pedalnetics, 2024-11-05:
Weird, definitely wasn't; Oh well, I'll just make that modification to the build file when I need to build it for whatever reason; the compile&firmware works just fine not using ninja so works for me