shanteacontrols/OpenDeck #310: Switch Debouncing

Pedalnetics, 2024-10-16:
Having an issue where often a switch sends 2 or more messages on press. I'm sure it's related to the contacts some-way, however my own Arduino code with debouncing does not suffer from this issue so I'm guessing that's why.

Is there a place/way to increase the debounce timing for Buttons? I did poke around the source and the isFiltered code without luck.

On a side note, and maybe my C is rusty, but doesn't this line here override the value passed via the function passed parameters?
https://github.com/shanteacontrols/OpenDeck/blob/f884b026be75194fe621e0353d257a980513ce63/src/firmware/application/io/buttons/Filter.h#L41

paradajz, 2024-11-27:
I'm sure back when I wrote that code it all made sense to me, but that is not the case anymore. I've rewritten the debouncer so you can check it out. The debounce time is now 8ms.

paradajz, 2024-11-28:
Released. https://github.com/shanteacontrols/OpenDeck/releases/tag/v7.11.0

closed 2024-11-28 by paradajz