shanteacontrols/OpenDeck #320: OpenDeck X(X)L Board Question

OhNoItSJoe, 2024-12-28:
Hello Shantea!

Upon searching for a MIDI-Controller to MIDIfy an old Organ to use it with Haupwerk and GrandOrgue I stumbled upon your project.

The organ I want to bring into the digital age has a "few" buttons tho.

[image]

183 keys, 25 Pedals, 60 Register buttons, 48 at the bottom and like 40 more, don't have it with me, yet.

Since I need more inputs, I'd like if its possible to simply increase the size of the shift registers of the L board to get more inputs. My estimations are 5-6 registers in total instead of the existing 2.

If I understood your circuit right, it would be as simple as adding more 74HC165 and daisy-chain them.

Since I also need more LEDs (I want to add a light to each key), I'd also have to quadruple the MAX2719, as far as I understood it's quite simple here, too since it also supports daisy-chaining.

Is this possible or am I forgetting something, maybe there are limitations of the Pico?

In theory I could also use 4 L boards for my project but I'd rather have one board.

OhNoItSJoe, 2024-12-28:
I've updated the schematics to represent the changes I'd like to do for my purposes.
Would this work?

Since I'm new to KiCAD, if someone wants to help me with setting up the PCB, help is appreciated.
Of course you'd get a sample of the board once I got it :)
[opendeck_xxl.zip](https://github.com/user-attachments/files/18267484/opendeck_xxl.zip)

laenzlinger, 2025-01-12:
what kind of help do you need?