shanteacontrols/OpenDeck #339: Impossible to build firmware

albertomaniac, 2025-06-25:
When in visual studio code i click on reopen in container, visual studio code gives me an error.

 "A mount config is invalid. Make sure it has the right format and a source folder that exists on the machine where the docker daemon is running"

I tried reinstall the virtual machine with ubuntu 22.04 and 24.04.

i've followed all the steps in the build guide and i don't know what i'm doing wrong.

i've tried also using 7.11.0 instead of 7.11.1 but in this case i can reopen container and got terminal in visual studio code but when i do 

 "make TARGET=pico"

i got the next error :

Generating` CMake files
Generating configuration for MCU: stub
Generating configuration for MCU: rp2040
Generating project-specific configuration for MCU: rp2040
Generating configuration for target: pico
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
fatal: not a git repository (or any parent up to mount point /home/ubuntu/workspace)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: not a git repository (or any parent up to mount point /home/ubuntu/workspace)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: not a git repository (or any parent up to mount point /home/ubuntu/workspace)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Configuring done (5.1s)
-- Generating done (0.1s)
-- Build files have been written to: /home/ubuntu/workspace/OpenDeck/build/pico/release
[252/271] Linking CXX executable bootloader.elf
Memory region         Used Size  Region Size  %age Used
             RAM:      186028 B       256 KB     70.96%
           FLASH:       32680 B        64 KB     49.87%
       SCRATCH_X:           0 B         4 KB      0.00%
       SCRATCH_Y:           0 B         4 KB      0.00%
[255/271] Performing configure step for 'sysexgen'
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (0.8s)
-- Generating done (0.0s)
-- Build files have been written to: /home/ubuntu/workspace/OpenDeck/build/pico/release/sysexgen
[256/271] Performing configure step for 'flashgen'
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
[... 47 lines cut ...]
                 from /home/ubuntu/workspace/OpenDeck/src/firmware/application/util/cinfo/CInfo.h:22,
                 from /home/ubuntu/workspace/OpenDeck/src/firmware/application/system/System.h:24,
                 from /home/ubuntu/workspace/OpenDeck/src/firmware/application/system/System.cpp:19:
/home/ubuntu/workspace/modules/SysExConf/include/SysExConf/SysExConf.h:222:18: note: candidate: 'void SysExConf::DataHandler::CustomResponse::append(uint16_t)'
  222 |             void append(uint16_t value)
      |                  ^~~~~~
/home/ubuntu/workspace/modules/SysExConf/include/SysExConf/SysExConf.h:222:18: note:   candidate expects 1 argument, 0 provided
/home/ubuntu/workspace/OpenDeck/src/firmware/application/system/System.cpp:426:30: error: no matching function for call to 'SysExConf::DataHandler::CustomResponse::append()'
  426 |         customResponse.append(SW_VERSION_MINOR);
      |         ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
/home/ubuntu/workspace/modules/SysExConf/include/SysExConf/SysExConf.h:222:18: note: candidate: 'void SysExConf::DataHandler::CustomResponse::append(uint16_t)'
  222 |             void append(uint16_t value)
      |                  ^~~~~~
/home/ubuntu/workspace/modules/SysExConf/include/SysExConf/SysExConf.h:222:18: note:   candidate expects 1 argument, 0 provided
/home/ubuntu/workspace/OpenDeck/src/firmware/application/system/System.cpp:427:30: error: no matching function for call to 'SysExConf::DataHandler::CustomResponse::append()'
  427 |         customResponse.append(SW_VERSION_REVISION);
      |         ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/workspace/modules/SysExConf/include/SysExConf/SysExConf.h:222:18: note: candidate: 'void SysExConf::DataHandler::CustomResponse::append(uint16_t)'
  222 |             void append(uint16_t value)
      |                  ^~~~~~
/home/ubuntu/workspace/modules/SysExConf/include/SysExConf/SysExConf.h:222:18: note:   candidate expects 1 argument, 0 provided
ninja: build stopped: subcommand failed.
make: *** [Makefile:37: all] Error 1

I hope someone could help me

paradajz, 2025-07-01:
I don't know what you're doing wrong. I've rebuild the container on my side and everything worked.

albertomaniac, 2025-07-01:
Hi, thanks for the reply. 
Could it be something wrong with my Visual Studio Code or Docker installation?

 I installed Ubuntu 22.04 with VirtualBox, Docker-CE with sudo permissions, and Visual Studio Code with the Remote Development extension.
 Is there anything else I need to install?

 Could it be that I'm having issues because I'm using a virtual machine?

Thanks.

closed 2025-07-01 by albertomaniac